esp-rs/esp-hal #3014: Wifi regression on esp32-c6

kaspar030, 2025-01-22:
## Bug description

We're seeing a wifi regression on esp32-c6, where the device seems to delay or drop packets it should receive.

It shows i.e., with weird ping times:

```
❯ ping 192.168.1.224                 
PING 192.168.1.224 (192.168.1.224) 56(84) bytes of data.               
64 bytes from 192.168.1.224: icmp_seq=4 ttl=64 time=3818 ms       
64 bytes from 192.168.1.224: icmp_seq=5 ttl=64 time=2805 ms
64 bytes from 192.168.1.224: icmp_seq=6 ttl=64 time=1791 ms
64 bytes from 192.168.1.224: icmp_seq=8 ttl=64 time=3875 ms
64 bytes from 192.168.1.224: icmp_seq=9 ttl=64 time=2851 ms
64 bytes from 192.168.1.224: icmp_seq=10 ttl=64 time=1838 ms
64 bytes from 192.168.1.224: icmp_seq=11 ttl=64 time=841 ms
```

We've seen this starting with 0.22.0 already. 0.21.1 works fine:

```
4 bytes from 192.168.1.224: icmp_seq=100 ttl=64 time=2.45 ms
64 bytes from 192.168.1.224: icmp_seq=101 ttl=64 time=10.8 ms
64 bytes from 192.168.1.224: icmp_seq=102 ttl=64 time=9.07 ms
64 bytes from 192.168.1.224: icmp_seq=103 ttl=64 time=8.51 ms
64 bytes from 192.168.1.224: icmp_seq=104 ttl=64 time=16.2 ms
64 bytes from 192.168.1.224: icmp_seq=105 ttl=64 time=6.67 ms
64 bytes from 192.168.1.224: icmp_seq=106 ttl=64 time=4.16 ms
64 bytes from 192.168.1.224: icmp_seq=107 ttl=64 time=14.1 ms
64 bytes from 192.168.1.224: icmp_seq=108 ttl=64 time=1.31 ms
64 bytes from 192.168.1.224: icmp_seq=109 ttl=64 time=18.3 ms
64 bytes from 192.168.1.224: icmp_seq=110 ttl=64 time=9.64 ms
64 bytes from 192.168.1.224: icmp_seq=111 ttl=64 time=8.80 ms
```

I tried bisecting, which leads to this commit:

```
fbc57542a8f4b71e30f0dcea4045c508ce753139 is the first bad commit                                                                                               
commit fbc57542a8f4b71e30f0dcea4045c508ce753139 (HEAD)                                                                                                         
Author: Dániel Buga <bugadani@gmail.com>                                                                                                                       
Date:   Tue Nov 12 11:36:25 2024 +0100                                                                                                                         
                                                                                                                                                               
    Remove pins from `Io` (#2508)                                              
     
    * Split pins off of Io
     
    * Remove the GPIO peripheral
     
    * p.GPIO

```

(... but that doesn't make sense to me)

## To Reproduce

The `wifi_embassy_dhcp` easily reproduces this. While it always manages to connect and get the nice crab drawing, *pinging* the node shows the weird timings:

```
SSID="foo" PASSWORD="bar" cargo xtask run-example examples esp32c6 wifi_embassy_dhcp
```

## Expected behavior

## Environment

- Target device: [e.g. ESP32-S3] 
```
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:51:39:b4
```

(I have esp32, esp32-s3 and esp32-c3 that seem to work fine.)

- Crate name and version: [e.g. esp-hal 0.20.0]

see above

Let me know how I can help debugging this!

bugadani, 2025-01-22:
Can you please see if `wifi_controller.set_power_saving(PowerSaveMode::None)` changes anything? I guess in https://github.com/esp-rs/esp-hal/pull/2446 we changed the default from None to Minimum, which may increase latency

kaspar030, 2025-01-22:
> Can you please see if `wifi_controller.set_power_saving(PowerSaveMode::None)` changes anything? I guess in [#2446](https://github.com/esp-rs/esp-hal/pull/2446) we changed the default from None to Minimum, which may increase latency

That fixes the issue, thanks! 👍 

Pings look like this again:

```
❯ ping 192.168.1.224
PING 192.168.1.224 (192.168.1.224) 56(84) bytes of data.
64 bytes from 192.168.1.224: icmp_seq=1 ttl=64 time=21.5 ms
64 bytes from 192.168.1.224: icmp_seq=2 ttl=64 time=20.0 ms
64 bytes from 192.168.1.224: icmp_seq=3 ttl=64 time=18.8 ms
64 bytes from 192.168.1.224: icmp_seq=4 ttl=64 time=6.91 ms
```

I'll not close this as maybe the default is suboptimal for the c6?

mchodzikiewicz, 2025-04-07:
I am a little bit upset about `PowerSaveMode::Minimum` as a default - I was affected by this in my hobby project at the end of the last year and it made me put that project on the shelf only to find this issue now.

I had a problem with MQTT messages that either arrived tens of seconds late or to never arrived, changed the policy to None today and the problem is gone 🤕

I don't know about y'all, but I first care about having something working and only then I optimize - I'd suggest to maybe have a good energy consumption optimization guide rather than making such policies the default

PR #3364 (2025-04-11): Misc esp wifi
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [ ] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have added necessary  [... cut]
closed 2025-04-11 by JurajSadel