esp-rs/esp-hal #3016: Revise GPIO Output drivers

bugadani, 2025-01-22:
The points to think about:

- We may not need a separate OutputOpenDrain driver. In fact, for future proofing reasons, we might want to prefer encoding the output mode in the associated configuration.
- We might want to omit the initial level from the associated configuration, and just pass it to the constructor. Setting the initial level is important to avoid glitches, but during operation the output level is already controlled by the driver. Including it in the configuration would mean either of the following:
    - We have an alternative way of specifying the output level, which is not ideal
    - Or, we would have to ignore a configuration field, which is not ideal

MabezDev, 2025-01-22:
Maybe we can also see if we can make config application completely infallible, might be worth exploring. Looking at the esp-idf gpio API, it seems most (all?) the errors returned are `INVALID_ARG` where someone passes a pin that doesn't exist in (and maybe where a pin can't go into a specific mode like some inputs on the esp32).

Dominaezzz, 2025-01-22:
> Maybe we can also see if we can make config application completely infallible, might be worth exploring

Yes please, less errors to handle the better. Errors make it more difficult to write robust firmware.

bjoernQ, 2025-01-24:
> We might want to omit the initial level from the associated configuration, and just pass it to the constructor.

I agree - an initial level is special enough to _not_ have it in the config

closed 2025-01-27 by bjoernQ