esp-rs/esp-hal #3020: [RFC] config options for specific chips

MabezDev, 2025-01-23:
Following on from the discussion here: https://github.com/esp-rs/esp-hal/pull/3001#discussion_r1924041094

We might want a better solution than just documenting which cfg's are supported on each chip (the current solution).

We one quite large constraint when building this:

* we can't break builds when a config option is passed by the user, _but_ the current target doesn't support that option. Doing so would break multi chip project setups.

With this constraint in mind, we still need to signal that the option isn't supported.

One idea I had:

## (ab)use cargo cfg linting

Where a cfg isn't supported, we could choose _not_ to emit this line: https://github.com/esp-rs/esp-hal/blob/16af010e71020e802a30efc1038226a02a1ff3ba/esp-config/src/generate.rs#L392.

This would flag a warning in user code, that the cfg they're trying to use has no effect. Unfortunately I don't think we can control the error message here, so the message is quite vague - but maybe something we could document in an FAQ. 

I would appreciate some more input/ideas here :).

I'm adding the 1.0 blocker label, though we might be able to add this later if we go with the above solution (iirc, introducing warnings is not a breaking change).

bugadani, 2025-01-23:
Cargo doesn't print warnings that originate in an external crate, so I'm not sure if such a warning would be visible. And besides, I'd prefer not seeing warnings if my project happens to support two MCUs and I'm trying to set specific configurations for them. Even though a workaround exists, it's not entirely ideal that one is needed.

bjoernQ, 2025-01-23:
My initial idea to support this was a bit different

e.g.
```toml
[env]
ESP_HAL_CONFIG_PLACE_ANON_IN_RAM = "true"

# don't do it on C2
ESP32C2_ESP_HAL_CONFIG_PLACE_ANON_IN_RAM = "false"

ESP32C6_ESP_HAL_CONFIG_FLIP_LINK = "true"
ESP32H2_ESP_HAL_CONFIG_FLIP_LINK = "true"
```

i.e. optionally prefix a key with a chip makes it override the default or the non-specific setting

Downsides
- makes `esp-config` esp-specific (probably not a real problem)
- what about using `esp-config` in a crate which is agnostic to the target (we don't have s.th. like that I guess - and it probably won't hurt)
- can't apply an override to multiple chips (see the FLIP-LINK example)
- even if a config is chip specific we should emit `cargo:rustc-check-cfg=cfg({cfg_name})` (is this a problem?)
- could we run into a similar situation which made us introduce the `_CONFIG_` separator?

bjoernQ, 2025-01-29:
Another thing we could but probably don't want to do is reading our own file (e.g. `esp-config.toml`) and populating the env-variables.

Could look something like this
```toml
[esp-hal]
place_anon_in_ram = true

[esp32s3.esp-hal]
psram-mode = "octal"

[esp-wifi]
mtu = 1524
```

Users could still override it via `[env]` and by setting env-variables

bugadani, 2025-01-29:
> and populating the env-variables

What would be the mechanism to do this? I'm not sure if a crate's build script can see the whole project's config file, and we can't really set env variables for other crates (i.e. from the project's build.rs, we can't affect esp-hal - https://github.com/esp-rs/esp-hal/pull/2156#issuecomment-2353671770).

bjoernQ, 2025-01-29:
> > and populating the env-variables
> 
> What would be the mechanism to do this? I'm not sure if a crate's build script can see the whole project's config file, and we can't really set env variables for other crates (i.e. from the project's build.rs, we can't affect esp-hal - [#2156 (comment)](https://github.com/esp-rs/esp-hal/pull/2156#issuecomment-2353671770)).

sure, we can't set env-vars for other crates but we can for the "to be compiled" crate which is enough - i.e. a crate would only pick it's own config keys

> crate's build script can see the whole project's config file

Would be the same mechanism used by `toml-cfg`

bugadani, 2025-01-29:
I guess it could work. I wouldn't see any reason (except that we now build on env vars in the HIL tests) to keep using env vars (at least as a user), but I guess that's not necessarily a big negative, just a bit of complexity we'd keep around.

MabezDev, 2025-04-04:
I'd like to avoid an `esp-config.toml` file for a number of reasons:

- I don't like more than one way of configuring
  - What gets precedence, the file or the env?
- As with toml_cfg it gets weird when using `--target-dir`
- Another thing we need to document

I'm leaning towards the chip prefix override at the moment. I'm not seeing any obvious ways it can go wrong.

> can't apply an override to multiple chips (see the FLIP-LINK example)

I'm not sure I understand this, in your example you are changing the settings for two chips? Did you mean multiple as in multiple at a time?

> even if a config is chip specific we should emit cargo:rustc-check-cfg=cfg({cfg_name}) (is this a problem?)

I think yes, but it shouldn't be a problem

bugadani, 2025-04-04:
> Another thing we need to document

You can add "We can process the .cargo/config.toml file equally easily" I think. We just need to be mindful to not mess with variables that don't belong to esp-config, but the tool doing that will need to know the available options anyway.

bjoernQ, 2025-04-07:
> > can't apply an override to multiple chips (see the FLIP-LINK example)

> I'm not sure I understand this, in your example you are changing the settings for two chips? Did you mean multiple as in multiple at a time?

Yeah, what I meant was we would need one variable for each to only set it for C6 and H2.

e.g.
```
ESP32C6_ESP_HAL_CONFIG_FLIP_LINK = "true"
ESP32H2_ESP_HAL_CONFIG_FLIP_LINK = "true"
```

I don't think it's a big deal - if we want some setting for all but one or two chips we would set the un-prefixed variable for all and then add the exceptions

MabezDev, 2025-05-09:
I think we should try the chip prefix override method from @bjoernQ. I don't see any obvious way this can go wrong, and it's the simplest, which I like. There are a couple of unsolved questions, but I think it's mostly a case of figuring out the most sane behavior rather than an inherent flaw with a particular approach.

To close this we should do the following:

1) Implement the proposal
* We'll need to expand `esp_config::ConfigOption` to contain a list of chips that the option supports (defaulting to all) (this might be doable with the `active` flag, not sure, I'll leave to the implementor.)
* Error if we detect a `$CHIP_$PACKAGE` combo that isn't applicable, this will require more thought.
  i.e `ESP32_ESP_HAL_CONFIG_FLIP_LINK = "true"` should be rejected outright but there are edge cases, like what should happen here:

  ```toml
  ESP_HAL_CONFIG_FLIP_LINK = "true"
  ESP32_ESP_HAL_CONFIG_FLIP_LINK = "false"
  ```
  Perhaps if a setting isn't supported by all chips, it _must_ get prefixed by a supported chip? For example `ESP_HAL_CONFIG_FLIP_LINK` is invalid because it's only supported on the C6/H2, therefore the only way to enable this setting is to write:

  ```toml
  ESP32C6_ESP_HAL_CONFIG_FLIP_LINK = "true"
  ESP32H2_ESP_HAL_CONFIG_FLIP_LINK = "true"
   ```

* Including copius tests to ensure we don't get this wrong
2) Find the right place to teach this override mechanism
  * We can document this in one place and refer to it
  * We can include the docs for this in esp-config's generator output

bugadani, 2025-05-09:
I think the best we could do here is **nothing**. `cargo` can use different config files (with `cargo build --config .cargo/configX.toml`), and so users can define different config files for different project configurations. This way the users aren't limited to one configuration per chip model, but any number.

This problem comes into play when the user either targets multiple chips, or the same chip with different feature sets. In either case they have some way to select between the configurations, either a cargo alias or a feature flag, or an xtask subcommand. So the way I suggest this problem be solved is: add `--config .cargo/specific_config.toml` to the cargo alias (create an alias if you don't use them already, they are convenient shortcuts), or the invocation in the xtask.

> As an aside, aliasses work in a strange way. Even if you specify `--config`, cargo will read the list of aliases from the default `.cargo/config.toml`. One one hand, this means they don't need to be duplicated for the different config files. On the other, what the heck.

Suggestions for esp-config TUI
------------------------------

esp-config should be as convenient as possible. When running `esp-config`:
- If there is a single cargo config file, just use it
- If there are multiple, and the user has specified one with `--config`, just use it
- If there are multiple and the user has not specified `--config`, display an interactive list and let the user select. I don't think a non-interactive option makes much sense, but it can be added if needed for some subprocess magic. In that case, esp-config should print the list of config files it found, then exit.

esp-config should try to guess the chip from the contents of the config file. We can help it by defining `ESP_HAL_TARGET`, which can then be used to further validate during build.
- If ESP_HAL_TARGET is not set, the config tool can try to filter out chips based on the selected config options.
  - If multiple chips match the config set, esp-config should display an interactive list and let the user select. `--non-interactive` could be provided as per above, in which case we could add `--chip` to the CLI, too. This can be used to further validate the config: we can warn the user if `--chip` does not match `ESP_HAL_TARGET`. We can then either overwrite ESP_HAL_TARGET or bail.
  - If no chips match, we can offer the user a recovery option: We can walk back on the filtering steps, and offer to remove the one(s) that remove the last chip from the list of possible matches. Alternatively, we could ask the user to manually fix the config file.
- It is possible that the config file contains invalid entries. esp-config should be able to display these errors and offer to reset them. Alternatively, we can just print the errors and bail.

If esp-config is used to semi-automatically fix issues, it should create a backup file so the user has a way to revert even without git.

Alternatives considered
-----------------------

- The above per-chip prefix/suffix solution. This is, in my opinion, inflexible as it limits to one configuration per chip model. Judging by my example (card/io has two separate hardware versions designed with an ESP32-S3 in the same project), this may be an actually problematic limitation.
- A more friendly DSL to describe the configuration. This can be encoded as a single, complex `ESP_CONFIG` environmental variable, and we can then include any feature we want in that language. This solution is complex, and doesn't help the esp-config TUI to pick the "active" configuration.
- A separate file, in TOML or other format. We could encode chip/feature-dependent values in a very flexible way, as for example `[esp-hal.cfg(feature = "foo").esp32s3]`. Assuming we can get env variables out of this, this also doesn't help the TUI, and the TUI wouldn't have an easy way to write back the perhaps complex configuration selector correctly.

I acknowledge that in both the TOML and the custom DSL cases we can teach the TUI to ask similar questions as I outlined in my suggestions. However, above we have at most 2: which file to use, which chip to use - but with more complex options mixed in a single files, the tool may ask more questions which can get annoying. Specifying a whole list of options because the format is flexible is not user-friendly.

bjoernQ, 2025-05-20:
After thinking (and forgetting) about this I think

> the best we could do here is nothing

is something I like. We could (and should) explain this in the advanced section of the book - and also define naming conventions etc.

bjoernQ, 2025-06-13:
> We made a discovery regarding `--config`, which may be interesting: [#3636 (comment)](https://github.com/esp-rs/esp-hal/pull/3636#discussion_r2145068416)

AND we should have something in the book about that (for the esp-config stuff)

bjoernQ, 2025-07-03:
I explored the idea of individual config.toml files here: https://github.com/bjoernQ/esp-hal-multiconfig-example

It works really, really nice IMHO

MabezDev, 2025-07-03:
> I explored the idea of individual config.toml files here: [bjoernQ/esp-hal-multiconfig-example](https://github.com/bjoernQ/esp-hal-multiconfig-example)

Thanks for this, it's really nice to be able to play around with the concept. We can also use this a basis for explaining the concept in the book.

I played around with this locally, and it really does work well. There are some changes I'd make to `esp-config`, like if it detects more than one `.toml` file in `.cargo` it would open a selection dialogue, etc, but in general it really does work well.

From one of the READMEs:

> To avoid confusion it's best to keep the whole [env] section in the individual config files.

I agree, and I think we should change esp-generate to enforce this. We should keep the base `.cargo/config.toml` for compilation/build options only, then keep our configs separate.

To summarize, I consider this issue solved. We have a good path forward. To complete the issue, I will:

* File an issue under esp-generate to use a separate config for our variables.
* File an issue for esp-config improvements.
* Add to a section of the book (added a note in https://github.com/esp-rs/book/issues/178)

If I've missed something, please let me know! Closing for now.

closed 2025-07-03 by MabezDev