esp-rs/esp-hal #3022: Inconsistent `#[unstable]` attribute application

MabezDev, 2025-01-23:
First discovered here: https://github.com/esp-rs/esp-hal/pull/3021#discussion_r1927062137

Sometimes we mark an impl block as unstable which produces some not so helpful docs:

[image]

Sometimes we apply it per function:

[image]

bugadani, 2025-01-23:
For this issue to be actionable, you should probably define what we should do. It's implied, but may not be clear or it may imply different things to us than to you :)

bugadani, 2025-01-23:
Impl blocks are tricky because they are simply removed if the feature isn't enabled. This pretty much is the only way to treat trait impls - there is no such thing as a private impl, or a trait impl with private methods.

We could maybe? redefine the attribute on inherent impl blocks, to instead apply the attribute to all items in that block. As this is pretty much my fault/mistake/oversight I don't think the ratatui team would mind, but I think it would be a weird (and breaking) change to handle different impl blocks in such a different way.

It's also easier to remove/stabilize instability markers if they are placed individually. It's also probably more obvious from reading the code what's stable and what's not.

TLDR; I think we should probably mark each function.

MabezDev, 2025-01-24:
> TLDR; I think we should probably mark each function.

I agree. We should also document this in API-GUIDELINES.

bugadani, 2025-01-27:
I don't understant why this would be a beta blocker, to be honest. As long as the unstable types can't be accessed, this is a docu/maintainability issue.

MabezDev, 2025-01-28:
> I don't understant why this would be a beta blocker, to be honest. As long as the unstable types can't be accessed, this is a docu/maintainability issue.

I added it because it's confusing, they might think parts of it are stable and then run into issues when they try to build it. Imo it's worth taking a bit of time to sort this now.

closed 2025-01-30 by jessebraham