esp-rs/esp-hal #3032: LCD peripheral is disable in esp-hal v0.23.0

orsinium, 2025-01-25:
## Bug description

After upgrading from esp-hal v0.22.0 to v0.23.0, the code using LCD stopped working: there are no errors reported but the screen stays inactive. After bisecting esp-hal, I found that the cause of the issue is #2544. Adding the following line of code at the beginning of `LcdCam::new` solves the issue:

```rust
crate::system::PeripheralClockControl::enable(crate::system::Peripheral::LcdCam);
```

Diff:

```diff
diff --git a/esp-hal/src/lcd_cam/mod.rs b/esp-hal/src/lcd_cam/mod.rs
index 0252f611..8214f2bf 100644
--- a/esp-hal/src/lcd_cam/mod.rs
+++ b/esp-hal/src/lcd_cam/mod.rs
@@ -36,6 +36,7 @@ impl<'d> LcdCam<'d, Blocking> {
     pub fn new(lcd_cam: impl Peripheral<P = LCD_CAM> + 'd) -> Self {
         crate::into_ref!(lcd_cam);
 
+        crate::system::PeripheralClockControl::enable(crate::system::Peripheral::LcdCam);
         let lcd_guard = GenericPeripheralGuard::new();
         let cam_guard = GenericPeripheralGuard::new();
```

## Environment

- Target device: ESP32-S3, ili9488.
- Crate name and version: esp-hal 0.23.1

Dominaezzz, 2025-01-25:
This should be fixed by #3007

Dominaezzz, 2025-01-25:
Re-open if `main` doesn't work for you as is.

closed 2025-01-25 by Dominaezzz