esp-rs/esp-hal #3034: I2C master: Read data does not not match actually present on bus.

katyo, 2025-01-26:
## Bug description

The data in read buffer does not match data which slave writes to the bus. For example it reads `0x83` while on a bus I see the `0xC3`, it reads `0x03` while on a bus `0x83` is present.
I deals with a `ESP32-C3`. I use the default `I2C` config with `SCL` on a `GPIO3`, `SDA` on a `GPIO4`. The hardware pull-ups is a 4.7K both. If that matter the slave device is a relative humidity and temperature sensor `HTU21D`.

[image]

[image]

Looks like the 7th and 6th bits changes randomly.

## To Reproduce

I use the following code snippet:

```rust
    esp_println::logger::init_logger_from_env();

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    let mut i2c = I2c::new(peripherals.I2C0, i2c::Config::default())
        .unwrap()
        .with_sda(peripherals.GPIO4)
        .with_scl(peripherals.GPIO3);

    loop {
        use embedded_hal::i2c::I2c;

        let mut d = [0u8];

        i2c.write_read(0x40, &[0xe7], &mut d).unwrap();
        println!("data: {d:x?}");

        Timer::after_secs(1).await;
    }
```

I tried both blocking and async modes with same results. I'm on a main `esp-hal` branch.

## Expected behavior

Read data same as an actually present on a bus.

## Environment

- Target device: ESP32-C3
  ```
  Chip type:         esp32c3 (revision v0.3)
  Crystal frequency: 40 MHz
  Flash size:        4MB
  Features:          WiFi, BLE
  MAC address:       60:55:f9:2b:8c:73
  ```
- Crate name and version:
  `esp-hal v0.23.1`

katyo, 2025-01-26:
If I lower the frequency to 50 kHz, the data is read correctly. It seems to be a hardware issue. May be it needs some configuration for input filters, but seems it still hasn't supported.

bugadani, 2025-01-27:
Thanks for reporting this! Just to confirm, the wavefor doesn't change in a meaningful way when you look at it, right?

bugadani, 2025-01-27:
Can you please check if https://github.com/esp-rs/esp-hal/pull/3040 solves your issue?

lure23, 2025-01-29:
@katyo Interested to follow this. Just making sure you know about potential problems if using all these together:
- ESP32-C3
- probe-rs
- I2C bus, via `esp-hal`
?

The problem is that `probe-rs` occasionally needs to stop the MCU core from running (doesn't need to do that in same way on, say, ESP32-C6), and this can cause effects on time-critical things like I2C. Because of this, I moved my i2C project to use C6.

If you don't use `probe-rs`, just ignore this.

bugadani, 2025-01-29:
> The [problem](https://github.com/lure23/ZOO-esp32/blob/main/TROUBLES.md#esp32-c3-problem-with-certain-time-critical-hardware-situations-and-the-jtag-interface) is that probe-rs occasionally needs to stop the MCU core from running

This is only an issue if you use anything RTT-based, like defmt, but not true in general. I'm not sure if the I2C peripheral would be affected by this, though...

lure23, 2025-01-29:
> This is only an issue if you use anything RTT-based, like defmt, but not true in general. I'm not sure if the I2C peripheral would be affected by this, though...

Hmm.. so I likely should change the writing in the link I made. I'll try!  

For me, I2C was indeed affected; that's why I brought this up. But it only occurred because my VL53L5CX peripheral takes a longish (>64kB) transfer, to upload its firmware.

MabezDev, 2025-02-04:
@katyo could you try the linked PR when you get the chance?

lure23, 2025-02-21:
@bugadani  wrote:
> Do you remember what your error is? Can you try if configuring `BusTimeout::Disabled` fixes the issue with probe-rs?

I can certainly try!

So what you mean is to - in my code - run things with default settings and with `BusTimeout::Disabled`, and report? I can do this at the latest in March 8-9th.

The larger context is that, earlier, long I2C interactions seemed to bring the "bad" out, but recently (change is `esp-hal` changes, I would guess) the long firmware-downloading portions seem to work. Instead, the problems occurred after the firmware download but were deterministic.

I simply wish I2C gets stable and dependable on all the combinations I use. Currently, only one works. I'll have a fresh brain to look at this in March. Currently, I think I would see bugs even where there aren't any!!!

bugadani, 2025-02-21:
We are seeing reports where long I2C transfers, interrupted by esp-wifi, end up timing out, and this is just one of my (maybe bad) ideas to try. We'll figure it out eventually :)

MabezDev, 2025-04-23:
@katyo We've had a few more I2C robustness improvements, could you try again on main?

Zercerium, 2025-05-17:
I'm still unable to get the VL53L4CD working on main branch with a bus speed of 100khz. With 10khz the device works. sadly I have no oscilloscope to check the data on the bus. (ESP32S3 in use). A RPi4 has not problems with the same lib and 100khz bus speed.

MabezDev, 2025-06-27:
Closing due to inactively, please file a new issue if you are still experiencing issues.