esp-rs/esp-hal #3038: bt-hci dependency pulled when enabled defmt feature

DBLouis, 2025-01-27:
## Bug description

When `defmt` feature of `esp-wifi` crate is enabled, `bt-hci` is added to Cargo.lock even though `ble` feature is not enabled.

## To Reproduce

Add `esp-wifi` with `ble` feature disabled and `defmt` enabled.

## Expected behavior

`bt-hci` dependency should not be added if `ble` feature is disabled.

## Environment

- Target device: ESP32
- Crate name and version: esp-wifi 0.12.0

bugadani, 2025-01-27:
This is a cargo quirk - Cargo.lock may contain optional dependencies even if they are not activated.

closed 2025-01-27 by bugadani

DBLouis, 2025-01-27:
Ok, but this is not without consequences, I can see different behavior when bt-hci is present in Cargo.lock, even if I am not using bluetooth and `ble` feature is not enabled.

bugadani, 2025-01-27:
What difference in behavior can you see?

bugadani, 2025-01-27:
You are enabling logging, why would you expect the executable to remain unchanged?

> [WARN] - esp_wifi_internal_tx 12290

The error code is 0x3002, which is `ESP_ERR_WIFI_NOT_STARTED`. Not sure why this would be related to bt-hci, but you're seeing the log line because you're enabling `defmt`.

DBLouis, 2025-01-27:
Makes sense