esp-rs/esp-hal #3048: Using `flip-link` fails the linker step if `defmt` is used

bjoernQ, 2025-01-28:
- generate a ESP32-C6 project usng `esp-wifi/wifi`, `defmt` + `async`
- enable flip-link

`cargo build --release`

We end up with a too large stack causing linker errors.

Everything is fine without `defmt`

MabezDev, 2025-01-28:
Presumably this also happens with just a hal project, or is there something specific that esp-wifi does here?

bjoernQ, 2025-01-28:
Oh yes - just wrote this from a message on Matrix - actually we don't need anything but `defmt` + flip-link to make the linker step fail

bugadani, 2025-01-28:
Is this a new issue, or something that existed before https://github.com/knurling-rs/flip-link/issues/101 was resolved?

bjoernQ, 2025-01-28:
🙈 

ok - making sure `linkall.x` is the last linker script (i.e. having `defmt.x` _before_ seems to make it work) - makes kind of sense since when we calculate the remaining space for the stack we should probably include everything

bjoernQ, 2025-01-29:
User reported it's working when changing the order of the linker scripts. We already honor this in `esp-generate` so I guess we can close this here since we can't do anything about it, really

closed 2025-01-29 by bjoernQ