esp-rs/esp-hal #3058: Implement application description for ESP-IDF image format

SergioGasquez, 2025-01-29:
New versions of ESP-IDF bootloader expect to have application description, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/app_image_format.html#application-description.

Here's the macro in `esp-idf-sys`: https://github.com/esp-rs/esp-idf-sys/blob/master/src/app_desc.rs#L2

@bugadani suggested having a similar macro on `esp_hal::init` that could be configured with `esp-config`

MabezDev, 2025-01-29:
One thing we need to be careful about using esp-config is that we namespace those options properly so they're specific to esp-idf's bootloader and won't overlap with any relevant mcu boot options.

MabezDev, 2025-01-29:
Adding the 1.0 blocker as we need to solve this update the bootloaders in espflash for v4? Correct me if I'm wrong

ivmarkov, 2025-02-03:
> Adding the 1.0 blocker as we need to solve this update the bootloaders in espflash for v4? Correct me if I'm wrong

That's correct. To my understanding, the first bootloaders that do read `esp_app_desc_t` are the ones from 5.2.3 and 5.3.2. They are specifically reading the two new fields in `esp_app_desc_t` (which were pulled from the `reserved` space) and which are named:
- `min_efuse_blk_rev_full`
- `max_efuse_blk_rev_full`

They should be set to whatever MCU version range the app can deal with (or to 0 - 99 by default, if the app is compatible with all MCU versions)

ivmarkov, 2025-02-03:
There's also this new guy named `mmu_page_size` which appears in ESP-IDF bootloaders >= 5.4. I haven't looked into it yet.

ivmarkov, 2025-02-03:
Related: https://github.com/esp-rs/espflash/issues/741

MabezDev, 2025-04-23:
Implemented in a seperate crate.

closed 2025-04-23 by MabezDev