esp-rs/esp-hal #3060: [ESP32-C6] I2C writes get stuck, since commit `f9203d` - affects 0.23.{01} and latest `main`

lure23, 2025-01-29:
## Bug description

I came back to a project of mine, using the VL53L5CX distance sensor, and after updating `esp-hal` changes to 0.23.0, it always got stuck in I2C operations.

I narrowed down the reason to the commit in the title.

## To Reproduce

A bit lengthy instructions (hw and a driver download with click-through are needed) [are here](https://github.com/lure23/ZOO-esp32/issues/20).

## Expected behavior

I can continue developing the code, using latest `esp-hal` releases.

### Actual behavour

I am stuck at 0.22.0 release.

## Environment

- Target device: `ESP32-C6`
- Crate name and version: esp-hal 0.23.0, 0.23.1 and latest `main` (`bb1ad1`)

bjoernQ, 2025-01-30:
I happen to own one of those ToF sensors and have a very simple example using it here: https://github.com/bjoernQ/vl53l5-c2rust

That one still works fine for me. (in the repo it's using a C3, but verified it's works the same on C6)

Your code is unfortunately huge - if you can reduce it a minimal reproducer would help a lot.

The referenced commit belongs to #2544 which basically just "turns off" the peripheral when it gets dropped (just to give some direction to look into)

Hard to say if that commit actually breaks something for you or if it just reveals a problem lurking in the code already before (not saying it's the case)

lure23, 2025-01-30:
Thanks @bjoernQ 

I'll likely need to carry on with what works for me (0.22), but:

>That one still works fine for me. (in the repo it's using a C3, but verified it's works the same on C6)

It's good to know stuff works elsewhere (with the same sensor).

>Your code is unfortunately huge - if you can reduce it a minimal reproducer would help a lot.

That made me realize it... might be. 😬 The size is certainly not the goal - on the contrary. The aim is to bridge the gap for a newcomer to get sensors working, but... if it cannot be used as a replication... perhaps something is wrong.

Ideally, it'd be something akin to Arduino in easy-of-embracing.

>The referenced commit belongs to https://github.com/esp-rs/esp-hal/pull/2544 which basically just "turns off" the peripheral when it gets dropped (just to give some direction to look into)

My code keeps the same `I2c` all through the example, so no dropping. Weird. 

I'm looking through the commit, the current (`main`) code, and your working code. Perhaps I spot something.. In the mean time, I'll leave this issue open, if it's okay.

>Hard to say if that commit actually breaks something for you or if it just reveals a problem lurking in the code already before (not saying it's the case)

I agree. Just learning.

bjoernQ, 2025-03-04:
Mind trying #3199 to see if it actually fixes this issue?

lure23, 2025-03-04:
@bjoernQ Looks interesting.

While the issue (by its title) is ESP32-C6 specific, the PR seems aimed at ESP32 and -S2. Would you like me to test it on C6?

lure23, 2025-03-10:
I have yet to try this out with the ZOO-esp32 repo, as described earlier. I plan to do so within this week.

On another code base, I "suddenly" have no issues whatsoever with ESP32-C6; tested with 0.23.0, 0.23.1, and 1.0.0-beta0. So it looks good.

To check the effect of the PR, I will:

- use `esp-hal` commit `e598146`
- use commit `f58d295`

If latter works, the PR changed something. Will come back, but likely this was all just noise.

closed 2025-03-10 by lure23

akauppi, 2025-03-14:
summary (after closing): there was no observed change between the two commits above.

I still have the ESP32-C6 I2C behaviour somewhat shaky - whereas C3 is stable. But this issue belongs to bygones.