esp-rs/esp-hal #3075: esp32c6 really bad WiFi range compared to C program

jpdisco, 2025-01-30:
## Bug description

When I make program with C in arduino editor for seeed lab xiao esp32c6 wifi works very well connects instantly and has good range.

when I make program in rust with esp wifi it can take a long time to connect and the range is bad, putting my hand between esp and router stops pings.

## To Reproduce

I used the static ip examples from this repo (with and without embassy) and the range was still very bad.

I even try setting power save on wifi to NONE and it still is bad!

## Expected behavior

I think range should be like what the arduino C editor gives me.

## Environment

esp-hal version = "0.23.1"
- 
RISC-V Chip:
  IDCODE: 000000dc25
    Version:      0
    Part:         13
    Manufacturer: 1554 (Espressif Systems (Shanghai)  Co Ltd)
Xtensa Chip:
  IDCODE: 0000000000
    Version:      0
    Part:         0
    Manufacturer: 0 (Unknown Manufacturer Code)

jpdisco, 2025-01-30:
Also only when using this library my esp gets very hot!

bjoernQ, 2025-01-31:
> Also only when using this library my esp gets very hot!

That's probably because our examples run at max-CPU-speed by default

According to the schematic your board can switch between the in-built antenna and an U.FL connector.

[image]

I assume you are not doing anything special and the default is the U.FL connector (without an antenna attached) - so either connect an antenna or make sure you are using the build-in antenna

bjoernQ, 2025-01-31:
[image]

Your code actively sets GPIO3 to low and GPIO14 to low?

jpdisco, 2025-01-31:
Yes I added to my code:

```rust
    let mut x = Output::new(peripherals.GPIO3, Level::Low);
    x.set_high();

    Timer::after(Duration::from_secs(1)).await;

    let mut y = Output::new(peripherals.GPIO14, Level::Low);
    y.set_low();
```

No effect unfortunately, I can still block wifi with my hand. I did not even have to do this with my C code because I think internal antenna is default.

bjoernQ, 2025-01-31:
Yeah for the Arduino code you select your board and it initializes things accordingly. So, the big question is what else do they configure differently

jpdisco, 2025-01-31:
I tried a few different esp32c6 boards in the list including the dev board one at the top and they all failed to compile.

bjoernQ, 2025-01-31:
t.b.h. I never every used Arduino at all - but it would be even much more helpful if you could run an esp-idf example on that board and see if it works fine ( e.g. https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station )

ivmarkov, 2025-02-03:
Could it be related to the problem described in https://github.com/esp-rs/esp-hal/pull/3081 ?

@jpdisco In other words, a quick workaround might be to do:
```rust
controller
        .set_power_saving(esp_wifi::config::PowerSaveMode::None)
        .unwrap();
```

jpdisco, 2025-02-05:
I have already set powersave to none, it only reduced latency not range problems. When I have time I will try esp idf!

ProfFan, 2025-03-08:
I have this suspicion that the PHY init could be slightly different in different chips especially given that the blob for 3rdparty RTOSes given by Espressif is slightly different from the blob in esp-idf. Might be worth it to do a register write dump by intercepting the init done by IDF v.s. the init done by ESP-WIFI and compare, using GDB watchpoints. I encountered a similar problem with another chipset before, it turned out to be an LDO inside the chip being enabled too late, and that LDO is critical to RF calibration.