esp-rs/esp-hal #3077: stability of configuration options

MabezDev, 2025-01-31:
I can foresee us wanting to try out some configuration options, like in https://github.com/esp-rs/esp-hal/pull/3054 but don't want to commit to having it around forever.

This could be a simple as documenting the stability, but maybe we should prefix unstable configs to make it more clear? `ESP_HAL_CONFIG_UNSTABLE_PLACE_SPI_DRIVER_IN_RAM` or something like that?

bugadani, 2025-01-31:
We also have the option to gate configs by the unstable feature, it's just a matter of cfg-ing compiler_error! calls.

Dominaezzz, 2025-01-31:
It would be a good idea to also take into account how stabilisation would work.
If `UNSTABLE` is in the config name, when the config is stabilised, users should get a nice message saying "This config is now stable, use this instead" rather than "This config isn't recognised".

If it's gated by the unstable feature, where appropriate, users should get a nice message saying "This has been renamed to ...." or "The possible values have been changed to ....". This one might be more difficult to implement. Might also be weird for a config to silently become stable whilst also introducing breaking changes.

bugadani, 2025-01-31:
> Might also be weird for a config to silently become stable whilst also introducing breaking changes.

That's a good shout, but we should also generally try our best and not insta-stabilize a breaking change.

bjoernQ, 2025-01-31:
I guess we should _not_ rename something in the process of stabilizing it.

Hiding an unstable config option behind the unstable feature sounds good to me.

Given they are only available for unstable I don't see a need to warn about their usage - e.g. we also don't warn the user when calling an unstable function. When they get renamed or removed it's no different from changing any other unstable API item

MabezDev, 2025-04-04:
I think cfg'ing away unstable configs is a good idea, and it's better than trying to resolve this with just naming conventions.

There are a few constraints for the impl though:

* We should still be able to discover these configs
    * They need to have docs generated for them, and marked as unstable
    * Maybe how to mark them as unstable still uses a "_UNSTABLE_" naming convention? Which will get renamed when they are stable?
    * If we do the above, we need to worry about reporting that its now stable to users as @Dominaezzz mentioned
* The default setting should still be applied where applicable
* What to do when a user is setting a config in [env] but `unstable` is not enabled?
    * Can we make this obvious to the user, or shall we leave this as a "documentation" issue

bugadani, 2025-04-14:
I typo'd the "Closes" in #3365, this issue is done, except for the potential revision of each config option.

closed 2025-04-14 by MabezDev