esp-rs/esp-hal #3078: Does esp-hal-procmacros need to be stable also?

MabezDev, 2025-01-31:
As part of not having dependencies <1.0 in the public API.

I assumed this was a yes and was going to open a separate issue, but looking at the docs it seems like macro reexports work a bit differently?

Take a look at https://docs.esp-rs.org/esp-hal/esp-hal/0.23.1/esp32c6/esp_hal/#attributes clicking on any macro and rustdoc seems to think the macro comes from esp-hal itself. If this is the case, it's great because we really only have to worry about esp-hal in this case.

bjoernQ, 2025-02-03:
Yes looks like this for me

bugadani, 2025-02-03:
We probably want to make it clear that we don't intend to stabilize procmacros and directly depending on that crate is not supported. With that, we can keep the crate unstable, but we also can't really introduce breaking changes into the reexported macros, which needs some care on our end.