esp-rs/esp-hal #3079: esp-wifi: Coex almost not working on ESP32-C6

bjoernQ, 2025-01-31:
Referencing https://github.com/ivmarkov/rs-matter-embassy/issues/3

Run the `wifi_coex` example and ping it from another machine - this works great on e.g. ESP32-C2, ESP32-C3 and ESP32-S3 - but barely on ESP32-C6

Given it works fine on C2 it's probably unrelated to the fact it's using a Nimble based driver.

bjoernQ, 2025-01-31:
Seems like COEX is now also broken on ESP32

ivmarkov, 2025-01-31:
> Seems like COEX is now also broken on ESP32

I can confirm that later with `rsp-matter-embassy` but last time it did work for me in that provisioning worked on the esp32. Anyway, will know for sure later today.

bjoernQ, 2025-01-31:
@ivmarkov #3081 should resolve the issue on ESP32-C6 if you set the power-saving-mode to None (hopefully)

ivmarkov, 2025-01-31:
@bjoernQ 

I confirm that this "power boost" call:
```rust
    controller
        .set_power_saving(esp_wifi::config::PowerSaveMode::None)
        .unwrap();
```

did fix the issue on the esp32c6!

With that said, here's my dirty little secret: I actually noticed I don't even enable the `coex` feature on the `esp-wifi` driver (?!). 
Yet, with our without `coex`, I do need to switch to non-power-saving mode the Wifi adapter - for esp32c6 specifically.

Now, how does it even work without `coex` being enabled? 
Not sure, but perhaps that's because I actually do need "very little coex" so to say. Due to how the Matter protocol works:
- Initially is is BLE only;
- At the final BLE stages, the Matter controller asks the Wifi to (a) scan for networks (that's optional, not all controllers do it) and (b) to connect to a specific network
- Weirdly, both ^^^ seem to work even with "coex" disabled
- Once the Esp reports successful connection, the BLE connection is abandoned (the controller unsubscribes) and from there on, it is really Wifi only (as in, the controller tries to find the just-provisioned over BLE Matter Node using an mDNS query, and if successful - it hits the device on the Matter UDP port)

So to cut the long story short, I don't think I would be enabling `coex` for now, and therefore, #3081 is not strictly necessary for me, as the APIs are already available in non-coex mode. But yes, with coex enabled I confirmed the connectivity is just as bad, so I think your PR is actually correct!

bjoernQ, 2025-01-31:
Wow that's interesting! But I'm glad it works for you - thanks for the update
(Need to try that coex without coex - maybe it's different for other chips)

ivmarkov, 2025-01-31:
This Matter-style "connect" and "scan"-only "coex without coex" seems to work on all chips, but the c6 this needs power boost.
But I'll confirm again in the next hours/days.

bjoernQ, 2025-04-16:
Closing this in favor of #3393

closed 2025-04-16 by bjoernQ