esp-rs/esp-hal #3103: 2Mbps UART crashes on initialization

TravisThomp, 2025-02-05:
## Bug description

Setting the baud rate to 2Mbps causes the change_baud function to overflow during creation.

```
====================== PANIC ======================
panicked at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.23.1/src/uart.rs:2461:36:
attempt to multiply with overflow

Backtrace:

0x4200758c
0x4200758c - esp_hal::uart::Info::change_baud
    at ??:??
0x42002593
0x42002593 - esp_hal::uart::Uart<Dm>::init
    at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.23.1/src/uart.rs:1247
0x42002801
0x42002801 - tentacle_lamp::____embassy_main_task::{{closure}}
    at /home/travis/projects/travis_art/projects/devices/lamp_rewrite/src/bin/async_main.rs:66
0x4200be5a
0x4200be5a - embassy_executor::raw::SyncExecutor::poll::{{closure}}
    at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.7.0/src/raw/mod.rs:430
0x42001da9
0x42001da9 - esp_hal_embassy::executor::thread::Executor::run
    at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-embassy-0.6.0/src/executor/thread.rs:106
0x42001d4b
0x42001d4b - tentacle_lamp::__xtensa_lx_rt_main
    at /home/travis/projects/travis_art/projects/devices/lamp_rewrite/src/bin/async_main.rs:32
0x4200a273
0x4200a273 - Reset
    at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.18.0/src/lib.rs:82
0x40378782
0x40378782 - ESP32Reset
    at /home/travis/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.23.1/src/soc/esp32s3/mod.rs:165
```

The overflow occurs in the `change_baud` function during the calculation of the clk_div
```rust
let max_div = 0b1111_1111_1111 - 1;
let clk_div = clk.div_ceil(max_div * baudrate); //This line
```
## To Reproduce

Create a uart device with a baudrate of 2Mbps
```rust
let encoder_uart = Uart::new(
    peripherals.UART1,
    Config::default().with_baudrate(2_000_000),
)
.unwrap();
```

## Expected behavior

The function doesn't overflow.

## Environment

- Target device: esp32s3 (revision v0.1)
- Crate name and version: esp-hal 0.23.1

MabezDev, 2025-02-05:
Could you try https://github.com/esp-rs/esp-hal/pull/3104?

TravisThomp, 2025-02-06:
No longer getting an overflow error,

Thank  you :)

closed 2025-02-06 by MabezDev