esp-rs/esp-hal #3114: Cannot install esp-println

maelp, 2025-02-07:
I get this error with v0.13

```
error[E0412]: cannot find type `PhantomData` in this scope
   --> /Users/primet/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-println-0.13.0/src/lib.rs:474:22
    |
474 | type LockInner<'a> = PhantomData<&'a ()>;
    |                      ^^^^^^^^^^^ not found in this scope
    |
help: consider importing this struct
    |
6   + use core::marker::PhantomData;
    |

error[E0425]: cannot find value `PhantomData` in this scope
   --> /Users/primet/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-println-0.13.0/src/lib.rs:487:21
    |
487 |         let inner = PhantomData;
    |                     ^^^^^^^^^^^ not found in this scope
    |
help: consider importing this unit struct
    |
6   + use core::marker::PhantomData;
    |

Some errors have detailed explanations: E0412, E0425.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `esp-println` (lib) due to 2 previous errors
```

MabezDev, 2025-02-07:
Closed via https://github.com/esp-rs/esp-hal/pull/2979, we still haven't released esp-println yet though. Could you use the git dependencies for esp-hal in the meantime?

maelp, 2025-02-07:
Thanks, so you mean using a git hash?

maelp, 2025-02-07:
I did, with the latest commit, but now I have this, any idea?

```

error: mach-o section specifier uses an unknown section type
    |
note: instantiated into assembly here
   --> <inline asm>:772:14
    |
772 |     .section .DebugExceptionVector.text,"ax",@progbits
    |              ^

error: unknown directive
    |
note: instantiated into assembly here
   --> <inline asm>:775:5
    |
775 |     .type _Level6InterruptVector,@function
    |     ^

error: unrecognized instruction mnemonic, did you mean: asr, lsr, msr, sri, str?
    |
note: instantiated into assembly here
   --> <inline asm>:777:9
    |
777 |         wsr a0, EXCSAVE6 // preserve a0
    |         ^

error: unrecognized instruction mnemonic
    |
note: instantiated into assembly here
   --> <inline asm>:778:9
    |
778 |         call0 __naked_level_6_interrupt     // used as long jump
    |         ^

error: mach-o section specifier uses an unknown section type
    |
note: instantiated into assembly here
   --> <inline asm>:780:14
    |
780 |     .section .NMIExceptionVector.text,"ax",@progbits
    |              ^

error: unknown directive
    |
note: instantiated into assembly here
   --> <inline asm>:783:5
    |
783 |     .type _Level7InterruptVector,@function
    |     ^

error: unrecognized instruction mnemonic, did you mean: asr, lsr, msr, sri, str?
    |
note: instantiated into assembly here
   --> <inline asm>:785:9
    |
785 |         wsr a0, EXCSAVE7 // preserve a0
    |         ^

error: unrecognized instruction mnemonic
    |
note: instantiated into assembly here
   --> <inline asm>:786:9
    |
786 |         call0 __naked_level_7_interrupt     // used as long jump
    |         ^

rustc-LLVM ERROR: Global variable '__default_exception' has an invalid section specifier '.rwtext': mach-o section specifier requires a segment and section separated by a comma.
error: could not compile `xtensa-lx-rt` (lib) due to 499 previous errors
warning: build failed, waiting for other jobs to finish...
```

jessebraham, 2025-02-07:
Given the mention of mach-o in the error, looks like you are building for macOS and not targeting Xtensa.

maelp, 2025-02-07:
Interesting, because I'm doing `cargo +esp build`, not sure how to configure?
Here's my config.toml

```
[target.xtensa-esp32s3-none-elf]
runner = "espflash flash --monitor"
rustflags = [
  # GNU LD
  #"-C", "link-arg=-Wl,-Tlinkall.x",
  #"-C", "link-arg=-nostartfiles",

  # LLD
   "-C", "link-arg=-Tlinkall.x",
   "-C", "linker=rust-lld",
]

#[build]
#rustflags = ["-C", "link-arg=-nostartfiles", "-C", "link-arg=-Trom_functions.x"]

target = "xtensa-esp32s3-none-elf"
features = "esp32s3"

[unstable]
build-std = ["alloc", "core"]

```

here's my Cargo.toml

```
[dependencies]
# Schedules asynchronous tasks for completion
embassy-executor = { default-features = false, features = ["task-arena-size-32768"], version = "0.7" }

# Plain TCP connection
embassy-net = { default-features = false, features = ["tcp", "dhcpv4-hostname"], version = "0.6" }

# Enables TLS 1.3 connection
embedded-tls = { default-features = false, git = "https://github.com/drogue-iot/embedded-tls" }

# Allocates heap memory
esp-alloc = { default-features = false, version = "0.6" }

# For example, useful to print the stack call of a raised error
esp-backtrace = { default-features = false, features = ["esp32s3", "panic-handler", "println"], version = "0.15.0", git = "https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d" }

# Contains the main features and structures
esp-hal = { default-features = false, features = ["esp32s3"], version = "0.23.1", git = "https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d" }

# Glue between Embassy and ESP
esp-hal-embassy = { default-features = false, features = ["esp32s3", "executors"], version = "0.6.0", git = "https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d" }

esp-println = { git = "https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d", version = "0.13.0", features = ["esp32s3"] }

# WiFi connectivity 
#esp-wifi = { default-features = false, features = ["esp32s3", "esp-alloc", "wifi"], version = "0.12" }

fugit = "0.3.7"

# Randomness generator
rand = { default-features = false, features = ["std_rng"], version = "0.9" }

# Parses certificate files
rustls-pemfile = { default-features = false, version = "2.2" }

# Pins values in static memory
static_cell = { default-features = false, version = "2.1" }
xtensa-lx-rt = { version = "0.18.0", features = ["esp32s3"], git = "https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d" }

[package]
edition = "2021"
name = "esp32-demo"
version = "0.1.0"

# Tweak parameters to reduce binary size
[profile.release]
codegen-units = 1
debug = false
debug-assertions = false
incremental = false
lto = true
opt-level = 'z'
overflow-checks = false
panic = 'abort'
rpath = false
strip = "symbols"

```

do you spot what I'm doing wrong? I'm a bit at a loss haha

maelp, 2025-02-07:
The error begins this way if it helps

```
   Compiling xtensa-lx-rt v0.18.0 (https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d#235ee624)
error[E0583]: file not found for module `arch`
  --> /Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-backtrace/src/lib.rs:46:1
   |
46 | pub mod arch;
   | ^^^^^^^^^^^^^
   |
   = help: to create the module `arch`, create file "/Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-backtrace/src/arch.rs" or "/Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-backtrace/src/arch/mod.rs"
   = note: if there is a `mod arch` elsewhere in the crate already, import it with `use crate::...` instead

   Compiling esp-hal-embassy v0.6.0 (https://github.com/esp-rs/esp-hal.git#235ee624f8805a0697c25e34b387698d88ee036d#235ee624)
error[E0425]: cannot find function `backtrace` in module `crate::arch`
  --> /Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-backtrace/src/lib.rs:69:34
   |
69 |     let backtrace = crate::arch::backtrace();
   |                                  ^^^^^^^^^ not found in `crate::arch`

error[E0425]: cannot find value `RA_OFFSET` in module `crate::arch`
  --> /Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-backtrace/src/lib.rs:79:48
   |
79 |         println!("0x{:x}", addr - crate::arch::RA_OFFSET);
   |                                                ^^^^^^^^^ not found in `crate::arch`
```

jessebraham, 2025-02-07:
```toml
#[build]
```

The build section is commented out, so the target is not being applied most likely.

maelp, 2025-02-07:
wooops you're right, now I have an error in my code so it compiles

```
it::MillisDurationU64;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^

error: `#[panic_handler]` function required, but not found

error[E0308]: mismatched types
   --> src/main.rs:35:17
    |
35  |     delay.delay(fugit::MicrosDurationU64::micros(2000 * 1000));
    |           ----- ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `Duration`, found `Duration<u64, 1, 1000000>`
    |           |
    |           arguments to this method are incorrect
    |
    = note: `Duration<u64, 1, 1000000>` and `esp_hal::time::Duration` have similar names, but are actually distinct types
note: `Duration<u64, 1, 1000000>` is defined in crate `fugit`
   --> /Users/primet/.cargo/registry/src/index.crates.io-6f17d22bba15001f/fugit-0.3.7/src/duration.rs:12:1
    |
12  | pub struct Duration<T, const NOM: u32, const DENOM: u32> {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: `esp_hal::time::Duration` is defined in crate `esp_hal`
   --> /Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-hal/src/time.rs:236:1
    |
236 | pub struct Duration(InnerDuration);
    | ^^^^^^^^^^^^^^^^^^^
note: method defined here
   --> /Users/primet/.cargo/git/checkouts/esp-hal-4998981da6dbb007/235ee62/esp-hal/src/delay.rs:54:12
    |
54  |     pub fn delay(&self, delay: Duration) {
    |            ^^^^^
```

maelp, 2025-02-07:
Thank you indeed this works... now just have to figure out why my esp32-s3 device is not detected by espflash on MacOS M1...

```
[2025-02-07T12:17:50Z INFO ] Detected 2 serial ports
[2025-02-07T12:17:50Z INFO ] Ports which match a known common dev board are highlighted
[2025-02-07T12:17:50Z INFO ] Please select a port
[2025-02-07T12:17:58Z INFO ] Serial port: '/dev/cu.usbmodem1101'
[2025-02-07T12:17:58Z INFO ] Connecting...
Error: espflash::connection_failed

  × Error while connecting to device
  ╰─▶ Failed to connect to the device
  help: Ensure that the device is connected and the reset and boot pins are not being held down
```

maelp, 2025-02-07:
Weirdly esptool.py doesn't seem to detect the device, althouh I know I was able to flash it before using this tool

```
esptool.py flash_id
esptool.py v4.8.1
Found 2 serial ports
Serial port /dev/cu.usbmodem1101
Connecting...
/dev/cu.usbmodem1101 failed to connect: Could not configure port: (6, 'Device not configured')
Serial port /dev/cu.debug-console
Connecting......................................
/dev/cu.debug-console failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
```

any idea?

closed 2025-02-07 by maelp