esp-rs/esp-hal #3120: `esp-println` logs are too spacy

lure23, 2025-02-09:
If I log using `esp-println` and `esp-flash` on the command line, there seem to be extra newlines between each logged line. Is this expected?

```
8.331165 INFO Zone : 0, Status : 5, Distance : 89 mm

8.331312 INFO Zone : 1, Status : 5, Distance : 90 mm

8.331437 INFO Zone : 2, Status : 5, Distance : 93 mm

8.331557 INFO Zone : 3, Status : 5, Distance : 95 mm

8.331682 INFO Zone : 4, Status : 5, Distance : 91 mm

```

I study this in the [esp-println-vs-probe_rs](https://github.com/lure23/espflash-vs-probe-rs) repo

Dominaezzz, 2025-02-09:
What OS are you doing espflash on?

lure23, 2025-02-09:
> What OS are you doing espflash on?

It's an Ubuntu Linux VM. `espflash 3.3.0`

bugadani, 2025-02-09:
https://github.com/lure23/espflash-vs-probe-rs/blob/c5a8251c4377ed86637c91ebf160aa650ef16521/examples/basic.rs#L155

You're printing a `\n`.

lure23, 2025-02-09:
> https://github.com/lure23/espflash-vs-probe-rs/blob/c5a8251c4377ed86637c91ebf160aa650ef16521/examples/basic.rs#L155

Wow, that's kind of embarrasing!   

Not my code. Gosh. Thanks. Closing! 😇

closed 2025-02-09 by lure23