esp-rs/esp-hal #3125: Allow DMA buffer disassociation to prevent memory duplication

almindor, 2025-02-10:
## Motivations

I will be using esp32-c6 for this example since that's how I found out about this.

Right now, using DMA buffers with SPI requires passing them into the `SpiDma` struct in order to obtain a `SpiBus/SpiDevice` capable handle, using `.with_buffers()`. This means that drivers, such as display drivers (e.g. [mipidsi](https://github.com/almindor/mipidsi)) cannot buffer their draw operations directly into the DMA tx buffer, but need to essentially duplicate buffers and do an unnecessary copy.

## Solution

If `SpiDma` itself was an implementor of `SpiBus` such that it was aware of the buffer descriptors and metadata but would not hold the actual `&mut [u8]` buffer pointer, it'd allow drivers to obtain access to that buffer directly and avoid the copy and additional buffer allocation.

## Alternatives

It might be possible to "expose" the internalized buffer reference but that would require locking which is probably more expensive than the copy in this case.

## Additional context

Manual reimplementation of Spi does allow this usage pattern.

Dominaezzz, 2025-02-10:
To summarize, you're basically asking for something in between `SpiDma` and `SpiDmaBus`.

- `SpiDma` - base driver
- `SpiDmaBus` - `SpiDma` + descriptors + perfectly aligned/located buffer
- `<the thing you want>` - `SpiDma` + descriptors (no buffer)

This is basically what `SpiDma` looked like before https://github.com/esp-rs/esp-hal/issues/1785 .

Some kind of wrapper would work.
```Rust
struct MySpiWrapper {
    spi_dma: SpiDma,
    descriptors: &'static mut [DmaDescriptor],
}
```
([Here's an example of someone doing the same thing but for a different peripheral](https://github.com/liebman/esp-hub75/blob/73d7f7ed6019a00a96f4e4ba1c387943e19f32c8/src/parl_io.rs#L26))

By going this route, you sacrifice the chunking that `SpiDmaBus` does for you and the "it just works" nature of it, but in exchange you get the zero copy DMA.

almindor, 2025-02-10:
> By going this route, you sacrifice the chunking that `SpiDmaBus` does for you and the "it just works" nature of it, but in exchange you get the zero copy DMA.

Exactly, which is the main pain point for doing something like screen rendering, especially considering RAM constrained setups where you can "just" fit the framebuffer into RAM, but not much more. It's not just about 0 copy, it's also about deduplicating memory use.

Frostie314159, 2025-02-25:
I second this, since it could provide a real performance improvement for my use case. I tried emulating the behavior with `SpiDma`, but it's pretty ugly.

Dominaezzz, 2025-03-29:
After dealing with https://github.com/esp-rs/esp-hal/issues/3262, I'm thinking whatever utility that's made to solve this issue would be difficult to use, given it'll return an error if the buffer isn't aligned or isn't in DMA-capable memory.
A good example is [this code in mipidsi](https://github.com/almindor/mipidsi/blob/4e5610d798b328f67c2cb0de2dd0ade7698d242d/src/models/gc9a01.rs#L84-L87) that initialises the display, it just won't work because the command arguments would end up in flash, which can't be DMA-ed. (Also see https://github.com/esp-rs/esp-hal/issues/1532 for a similar problem).
Of course, if a user is willing to sign the terms & conditions for using it, it's still useful to have.

> If what we're doing in the Wi-Fi driver is unsound, please LMK

@Frostie314159 it's probably unsound if it's not blocking.

I'm guessing the Wi-Fi DMA doesn't care about alignment?

[This](https://github.com/esp32-open-mac/esp-wifi-hal/blob/291336082e63e97099bc8b14ebed745704c920ec/esp-wifi-hal/src/wmac.rs#L416) will have to be `'static`. See https://docs.rust-embedded.org/embedonomicon/dma.html for more info.

Frostie314159, 2025-03-29:
That's definitely a good point. I'm guessing that there probably aren't that many use cases for this anyway. 

Thanks for the pointer on DMA, I'll change the driver accordingly. Regarding the unsoundness. We use a drop guard to reset the transmission slot, if the Future is dropped. I'm by no means certain, that this makes it sound, but it's my best shot. The alignment does not matter for the Wi-Fi peripheral.

almindor, 2025-03-29:
> After dealing with [#3262](https://github.com/esp-rs/esp-hal/issues/3262), I'm thinking whatever utility that's made to solve this issue would be difficult to use, given it'll return an error if the buffer isn't aligned or isn't in DMA-capable memory. A good example is [this code in mipidsi]

Inits NOT the main use case for this. Nobody has display init code in their hot path. The use case is that draw operations would go through the DMA buffer and that is what we want. Init doesn't need to use DMA at all.

If safety is an issue I think this could be exposed as an `unsafe` API with documented caveats.

Dominaezzz, 2025-03-29:
> We use a drop guard to reset the transmission slot,

Drop guard doesn't work when mem::forget is used.

> Inits not the main use case for this. Nobody has display init code in their hot path.

It's not about main use case or hot/cold paths.
It's about the fact that people will try to use it and it would return errors.

> Init doesn't need to use DMA at all.

How is the driver supposed to know when init is happening or not? Spi is spi.

almindor, 2025-03-29:
> It's not about main use case or hot/cold paths. It's about the fact that people will try to use it and it would return errors.
> 

Disallowing use because of bad abstractions is worse than allowing dangerous use. I like rust and it's safety guarantees but people go overboard and we end up with embedded-hal and broken SPI abstraction for years.
If you feel strongly about it, hide it behind `unsafe`.

> 
> How is the driver supposed to know when init is happening or not? Spi is spi.

By driver do you mean the SpiDevice or the "driver" as in for example Mipidsi? In case of the latter, of course it knows what's what and can use its interfaces accordingly. E.g. commands wouldn't use the buffer, but drawing operations would. This is already distinct.

Dominaezzz, 2025-03-29:
> Disallowing use because of bad abstractions is worse than allowing dangerous use.

There's no danger here or even unsafe.
The DMA hardware will literally not work if the buffer isn't exactly how it wants it to be.

> By driver do you mean the SpiDevice or the "driver" as in for example Mipidsi?

I mean the former.

almindor, 2025-03-29:
> There's no danger here or even unsafe. The DMA hardware will literally not work if the buffer isn't exactly how it wants it to be.
>
>
> I mean the former.

I think we're talking a bit over each other here then. My suggestion expects that a single `SpiDevice` exposes a way to "write via DMA" and "write without DMA" without having to own the DMA buffer.

I'm not sure how to best abstract this though.

bugadani, 2025-03-29:
> My suggestion expects that a single SpiDevice exposes a way to "write via DMA" and "write without DMA" without having to own the DMA buffer.

In that case, the embedded-hal implementation will always work without DMA, which isn't ideal.