esp-rs/esp-hal #3127: Make configs public

Easyoakland, 2025-02-11:
## Motivations

For my particular case I have to use the `esp-wifi` `ESP_WIFI_CONFIG_MTU` in my code to statically allocate buffers of the correct size.

## Solution

Config values publicly exposed.

## Alternatives

Manually copy the needed values into user code. This option isn't great because of the duplication.

## Additional context
I'm not sure what advantages are enabled by making the config values private. If the configs change, that is semantically a breaking change even if the structs are private in code.

bugadani, 2025-02-11:
You should be able to access configuration using `esp_config`. See https://github.com/esp-rs/esp-hal/tree/main/esp-config#capturing-configuration-values-in-the-downstream-crate

Easyoakland, 2025-02-12:
It would be nicer to import the config values defined by the corresponding crate. Is there a reason that's not possible or not preferred?

Easyoakland, 2025-02-12:
> We can generate the config struct that holds all values (as e.g. a `esp_hal::ESP_CONFIG` constant). If you want to send a PR to do this, you are welcome to do so.

Ok, good to know.

> > If the configs change, that is semantically a breaking change even if the structs are private in code.
> 
> That is a funny observation. The user manages the configuration, so if, based on the configuration, something changes in a breaking way (like behaviour, or the aformentioned generated config struct), is that really a breaking change? I don't think so.

If you, for example, remove or rename a config, then even if the struct is private the user may be setting a config which is now invalid. My point was just that private stuff is normally to hide implementation details, but user-facing modifiable parameters are by definition not implementation details.

MabezDev, 2025-02-18:
As things currently stand, the way to obtain configuration values is to use the helpers in `esp-config`. Not every crates has it's config in one place like esp-wifi, so it's easier to just capture from the environment.

closed 2025-02-18 by MabezDev