esp-rs/esp-hal #3148: rust-analyzer can not recognize `peripherals.GPIO*`.

traversebitree, 2025-02-19:
## Bug description

rust-analyzer can not recognize `peripherals.GPIO*`. But the compilation was successful.

[image]
***
[image]

## To Reproduce

1. Generate codes using `esp-generate`
```sh
esp-generate --chip esp32s3 blinky
```
2. Modify `src/async_main.rs`. My LED is at GPIO#48.
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{
    delay::Delay,
    gpio::{Level, Output},
    main,
};

use esp_println::println;

#[main]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    println!("Hello world!");

    // Set GPIO7 as an output, and set its state high initially.
    let mut led = Output::new(peripherals.GPIO48, Level::Low);

    led.set_high();

    // Initialize the Delay peripheral, and use it to toggle the LED state in a
    // loop.
    let delay = Delay::new();

    loop {
        led.toggle();
        delay.delay_millis(500);
    }
}
```

## Expected behavior

I expect when I typed `peripherals.GP` in VS Code. Then `peripherals.GPIO48` can appear.

## Environment

- Target device: ESP32-S3]
```sh
PS C:\Users\xykon> espflash board-info
✔ Use serial port 'COM3' - USB 串行设备 (COM3)? · yes
✔ Remember this serial port for future use? · no
[2025-02-19T04:06:40Z INFO ] Serial port: 'COM3'
[2025-02-19T04:06:40Z INFO ] Connecting...
[2025-02-19T04:06:40Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        32MB
Features:          WiFi, BLE
MAC address:       84:fc:e6:7e:4e:74
```
- Crate name and version: `esp-hal = { version = "0.23.1", features = ["esp32s3", "unstable"] }`

bugadani, 2025-02-19:
[image] is the current state. Also while this was caused by our macros, the issue is really a rust-analyzer issue.

traversebitree, 2025-02-19:
Thank you very much for your answer, so how can I solve this problem by myself?Or wait for the new version to be released?

bugadani, 2025-02-19:
You can wait for a new release, or you can start using our git `main` branch for the time being.

traversebitree, 2025-02-19:
Thank you for your help in resolving my issue!

closed 2025-02-19 by traversebitree

bugadani, 2025-08-07:
Since things compile just fine, I'd suggest opening a rust-analyzer issue.

david-boles, 2025-08-11:
Should this issue be re-opened? I believe I'm running into the same issue.

david-boles, 2025-08-12:
Thanks! Cherry picking 64968bb436dfd709076da7b7cb6757ed15fb14f2 onto v1.0.0-rc.0 seems to have worked just fine.

joebnb, 2025-10-01:
i'm in here now, may i ask when will rc2 release?