esp-rs/esp-hal #3153: ESP32: Dual-SPI broken in multiple ways

bjoernQ, 2025-02-19:
Running the `spi_halfduplex_read_manufacturer_id` example gives wrong results for `Dual`

```
Single [c8, 16]
Dual [20, 5b]
Quad [c8, 16]
```

While I'm sure I tested this before I get the same results with older esp-hal version. (But note: I am using a different flash chip now since I fried the one I used for testing, but it works fine for e.g. ESP32-C3, ESP32-S2, ESP32-C6 so it's not generally the flash chip's fault)

Dominaezzz, 2025-02-19:
Might be time to improve the HIL tests, maybe we can use the slave driver to properly test the master driver.
I suppose this won't work for chips with just one SPI peripheral

bjoernQ, 2025-02-26:
one (maybe) interesting observation: ESP32 has `SPI_MISO_DELAY_NUM` and `SPI_MISO_DELAY_MODE` which affects what I get here. e.g. delay_mode = 1 I read `08 16` - then with dummy = 8 it reads `c8 16`

bugadani, 2025-03-03:
Just a fun observation, for me it's Quad-SPI that doesn't work:

```
Single [20, 17]
Dual [20, 17]
Quad [64, 57]
```

bugadani, 2025-03-03:
Turns out I can't reproduce the issue. The flash chip I mined out of an ESP32 WROVER-IE module works as expected with the `spi_halfduplex_read_manufacturer_id` example. Still I'd appreciate if you could retry with #3201 applied

bjoernQ, 2025-03-04:
> Turns out I can't reproduce the issue. The flash chip I mined out of an ESP32 WROVER-IE module works as expected with the `spi_halfduplex_read_manufacturer_id` example. Still I'd appreciate if you could retry with [#3201](https://github.com/esp-rs/esp-hal/pull/3201) applied

I get slightly different results now on ESP32
```
Single [c8, 16]
Dual [2, 28]
Quad [64, 44]
```

(It's still fine on e.g. C6)

bjoernQ, 2025-03-04:
Ok - now reading 6 bytes I am back to my original results (maybe I screwed up wiring before)
```
Single [c8, 16, c8, 16, c8, 16]
Dual [20, 5b, 20, 5b, 20, 5b]
Quad [c8, 16, c8, 16, c8, 16]
```

Using a dummy-phase of 3 gives `Dual [16, c8, 16, c8, 16, c8]` 

oh .... changing SCLK from 0 to e.g. 15 now reads the right thing: `Dual [c8, 16, c8, 16, c8, 16]`

bugadani, 2025-03-04:
Dual SPI shouldn't need any dummy bits 👀

bjoernQ, 2025-03-04:
I forgot I also changed my devboard from a generic no-name thing to a DevKitC (because espflash can't talk to that generic one anymore) .... used esptool to flash it and it works fine with GPIO0

closed 2025-03-05 by MabezDev