esp-rs/esp-hal #3167: Frequent I2C timeouts when running I2C alongside WiFi stack

petekubiak, 2025-02-22:
## Bug description

Bug was seen when driving a display with a flush mechanism (full screen redraw) at the same time as running the WiFi stack on ProS3. If the WiFi is not running, I2C timeouts are not seen.

## To Reproduce

Bug was observed on a ProS3 connected via I2C to an Adafruit FeatherWing display. The display driver used was https://github.com/aaron-hardin/sh1107
1. Connect display to 4-pin connector for I2C on the ProS3
2. Flash code which spins up the WiFi stack and also updates the display at regular intervals (e.g. 10 Hz)
3. Print I2C errors to trace
4. Observe regular timeouts, also visual glitches on the display

I will attach a minimal example once I have the opportunity to put it together.

## Expected behavior

No I2C timeouts caused by running alongside WiFi.

## Environment

- Target device: ProS3
- Crate name and version: esp-hal 0.23.1

katyo, 2025-02-23:
Link this issue: #3034

bjoernQ, 2025-02-24:
Problem seems to be that the scheduler will switch to another task while the I2C driver tries to catch-up filling the FIFO (which it needs to do in-time)

Here is some simplified code showing this effect w/o esp-wifi
```rust
//! Read calibration data from BMP180 sensor
//!
//! This example dumps the calibration data from a BMP180 sensor
//!
//! The following wiring is assumed:
//! - SDA => GPIO4
//! - SCL => GPIO5

//% FEATURES: esp-hal/unstable embassy
//% CHIPS: esp32 esp32c2 esp32c3 esp32c6 esp32h2 esp32s2 esp32s3
//% TAG: bmp180

#![no_std]
#![no_main]

use embassy_executor::Spawner;
use esp_backtrace as _;
use esp_hal::{
    delay::Delay,
    handler,
    i2c::master::{Config, I2c},
    time::Duration,
    timer::{systimer::SystemTimer, timg::TimerGroup, PeriodicTimer},
};
use esp_println::println;

#[esp_hal_embassy::main]
async fn main(_spawner: Spawner) -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    esp_alloc::heap_allocator!(size: 72 * 1024);

    let syst = SystemTimer::new(peripherals.SYSTIMER);
    esp_hal_embassy::init(syst.alarm0);

    let timg0 = TimerGroup::new(peripherals.TIMG0);

    let mut periodic = PeriodicTimer::new(timg0.timer0);

    periodic.set_interrupt_handler(handler);
    periodic.start(Duration::from_millis(10)).unwrap();
    periodic.enable_interrupt(true);

    // Create a new peripheral object with the described wiring and standard
    // I2C clock speed:
    let mut i2c = I2c::new(peripherals.I2C0, Config::default())
        .unwrap()
        .with_sda(peripherals.GPIO4)
        .with_scl(peripherals.GPIO5)
        .into_async();

    loop {
        let mut data = [0u8; 220];
        println!(
            "{:?}",
            i2c.write_read_async(0x77, &[0xaa; 64], &mut data).await
        );
    }
}

#[handler]
fn handler() {
    unsafe {
        let timg0 = TimerGroup::new(esp_hal::peripherals::TIMG0::steal());
        let mut periodic = PeriodicTimer::new(timg0.timer0);

        let mut _d = Delay::new();
        _d.delay_millis(1);

        periodic.clear_interrupt();
    }
}
```

In that code lowering the 64 bytes written to something that fits the FIFO should make the errors disappear. Removing the delay from the interrupt handler, too.

Looking at the driver here: https://github.com/aaron-hardin/sh1107/blob/26bb3020fa825ef706b303ae017911e0aeb71931/src/interface/i2c.rs#L46 it's using 64 byte chunks while the FIFO is 32 bytes - probably just lowering that might do the trick in your case

petekubiak, 2025-02-24:
I tried changing the chunk length in the driver to 32 but the errors still seem to occur at the same rate as before.

petekubiak, 2025-02-24:
As far as I can tell It hasn't made a difference. Maybe with the environment variable set it causes bursts of errors to occur more often? I wouldn't count on that observation though, as it's a very random occurrence so it would be very difficult to quantify.

Something I've noticed (unrelated to the env variable) is that it seems like the errors occur more frequently when more is happening with the WiFi. There tend to be a couple of bursts of errors initially when the stack is being spun up, then it calms down a bit, but when connection to an AP is established (the board is operating in STA mode), the errors seem to become more common. Probably not surprising, but thought it was worth mentioning anyway.

bjoernQ, 2025-02-24:
> it seems like the errors occur more frequently when more is happening with the WiFi

Would make sense - if there is nothing to do for the WiFi driver we quickly return to the main task

> I tried changing the chunk length in the driver to 32 but the errors still seem to occur at the same rate as before.

Had another look at the code - they take that 32 and add 1 to it the next line for the actual data buffer to be written - so 31 here might be worth a try

petekubiak, 2025-02-24:
Setting `CHUNKLEN` to 31 causes a panic due to mismatched slice lengths. It seems that the driver is very much not designed for the chunk to be anything other than 64 - when I set it to 32 it completely messed up the positioning of what was being drawn on the screen. I think it's sheer luck that that didn't also cause a panic tbh.

I'm sure the driver could be modified to properly handle different buffer lengths, but this would be an entirely separate endeavour.

petekubiak, 2025-02-24:
I've been working on a minimal example to exhibit what I've seen and it's led to some interesting observations. The key takeaway from what I've seen is that I can recreate the issue with a very small example which doesn't even require the wifi:

## Timeouts with `defmt`
```rust
#![no_std]
#![no_main]

use alloc::format;
use core::cell::RefCell;
use embassy_time::Timer;

use defmt::{info, warn};
use embassy_embedded_hal::shared_bus::blocking::i2c::I2cDevice;
use embassy_executor::Spawner;
use embassy_sync::{blocking_mutex::raw::NoopRawMutex, blocking_mutex::Mutex};
use embedded_graphics::{
    fonts::{Font6x8, Text},
    pixelcolor::BinaryColor,
    prelude::*,
    style::TextStyle,
};
use esp_hal::{
    clock::CpuClock,
    i2c::{
        self,
        master::{BusTimeout, I2c},
    },
    time::RateExtU32,
    Async,
};
use sh1107::mode::GraphicsMode;
use sh1107::prelude::*;
use static_cell::StaticCell;
use {defmt_rtt as _, esp_backtrace as _};

extern crate alloc;

static I2C: StaticCell<Mutex<NoopRawMutex, RefCell<I2c<'_, Async>>>> = StaticCell::new();

#[esp_hal_embassy::main]
async fn main(_spawner: Spawner) {
    // generator version: 0.2.2

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    esp_alloc::heap_allocator!(72 * 1024);

    let timer0 = esp_hal::timer::systimer::SystemTimer::new(peripherals.SYSTIMER);
    esp_hal_embassy::init(timer0.alarm0);

    info!("Embassy initialized!");

    let i2c_bus = I2C.init(Mutex::new(RefCell::new(
        I2c::new(
            peripherals.I2C0,
            i2c::master::Config::default()
                .with_frequency(400.kHz())
                .with_timeout(BusTimeout::BusCycles(1)),
        )
        .unwrap()
        .with_sda(peripherals.GPIO8)
        .with_scl(peripherals.GPIO9)
        .into_async(),
    )));
    let i2c = I2cDevice::new(i2c_bus);
    let mut display: GraphicsMode<_> = sh1107::Builder::new()
        .with_rotation(DisplayRotation::Rotate90)
        .with_size(DisplaySize::Display64x128)
        .connect_i2c(i2c)
[... 178 lines cut ...]
    display.init().unwrap();

    let mut counter = 0;
    let mut errors = 0;
    loop {
        display.clear();
        Text::new(
            format!("Counter: {counter}\nError count: {errors}").as_str(),
            Point { x: 0, y: 28 },
        )
        .into_styled(TextStyle::new(Font6x8, BinaryColor::On))
        .draw(&mut display)
        .unwrap();
        let _ = display.flush().inspect_err(|err| {
            errors += 1;
            warn!("Error: {}", format!("{err:?}").as_str())
        });

        counter += 1;
        Timer::after_millis(100).await;
    }
}
```

`defmt` seems to cause significantly more errors than using the WiFi: ~560 errors per 1000 frames, vs ~30 errors per 1000 frames when running with WiFi and `esp_println`.

bjoernQ, 2025-02-25:
For the WiFi case it _might_ be a workaround to wrap the flushing in a critical section like this
```rust
        critical_section::with(|_| {
            let _ = display.flush().inspect_err(|err| {
                errors += 1;
                warn!("Error: {}", format!("{err:?}").as_str())
            });    
        });
```

That's not a great solution - I wouldn't even call it a solution t.b.h. - but maybe WiFi will tolerate the latency resulting from blocking interrupts for a short time

petekubiak, 2025-02-25:
> For the WiFi case it _might_ be a workaround to wrap the flushing in a critical section like this
> 
>         critical_section::with(|_| {
>             let _ = display.flush().inspect_err(|err| {
>                 errors += 1;
>                 warn!("Error: {}", format!("{err:?}").as_str())
>             });    
>         });
> 
> That's not a great solution - I wouldn't even call it a solution t.b.h. - but maybe WiFi will tolerate the latency resulting from blocking interrupts for a short time

I thought I'd give it a try with my original project which actually does something with the wifi. This does stop the I2C errors, but it also seems to stop the application from being able to accept connections and serve web pages, which I guess isn't that surprising.

bjoernQ, 2025-03-04:
Mind trying #3199 to see if it actually fixes this issue?

closed 2025-03-06 by bugadani

petekubiak, 2025-03-06:
Ahh perfect, I knew I'd seen someone do this before but I couldn't remember where or how!

bugadani, 2025-03-06:
You can only apply a patch if the crate version matches. This means you'll need to update to 1.0.0-beta.0 first.

petekubiak, 2025-03-06:
Ok so there have been some fairly hefty changes to esp-wifi on `main` which aren't making it easy to test against the original project (I'll try to get to it eventually but it'll require a rework of the code).

However, testing against the `defmt` example I shared above, the fix completely eliminates the I2C timeouts. Nice work!

PR #3199 (2025-03-06): ESP32/ESP32-S2: I2C Workaround
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have [... cut]