esp-rs/esp-hal #3178: I2C timeouts reading ublox gps

yalcinozhabes, 2025-02-25:
## Bug description

I have a Sparkfun SAM-M10Q GPS module which outputs NMEA sentences on i2c port. The output is around 1Kbytes per second. When I read the data to a buffer of size 32 or more bytes, I only get timeouts. When I keep the buffer size 31, it works.

Probably related to #3034 and/or #3167.

## To Reproduce

```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{
    delay::Delay,
    i2c::master::{BusTimeout, Config, I2c},
    main,
};
use esp_println::{print, println};

const DATA_BUF_SIZE: usize = 32;

#[main]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());
    let delay = Delay::new();
    let mut i2c = I2c::new(
        peripherals.I2C0,
        Config::default().with_timeout(BusTimeout::Maximum),
    )
    .unwrap()
    .with_sda(peripherals.GPIO16)
    .with_scl(peripherals.GPIO17);

    loop {
        let mut size_buf = [0u8; 2];
        let mut data_buf = [0u8; DATA_BUF_SIZE];
        match i2c.write_read(0x42, &[0xfd], &mut size_buf) {
            Ok(_) => {}
            Err(err) => {
                println!("Error reading the size {:?}", err);
                delay.delay_millis(500);
                continue;
            }
        };
        let size: usize = ((size_buf[0] as u16) << 8 | (size_buf[1] as u16)).into();

        let mut remaining = size;
        println!(
            "Got size_buf {:?}: size:{:?}",
            size_buf, remaining
        );

        while remaining > 0 {
            let bytes_to_read = if remaining > DATA_BUF_SIZE {
                DATA_BUF_SIZE
            } else {
                remaining
            };

            match i2c.write_read(0x42, &[0xff], &mut data_buf[0..bytes_to_read]) {
                Ok(_) => {}
                Err(err) => {
                    println!("Error reading data: {:?}", err);
                    break;
                }
            }
            remaining -= bytes_to_read;
            print!("{}", core::str::from_utf8(&data_buf).unwrap());
        }
        println!("Finished reading GPS data");
        println!("");
        delay.delay_millis(500);
    }
}
```
For `DATA_BUF_SIZE` > 31, I just get timeouts.
```
Got size_buf [0, 0]: size:0
Finished reading GPS data

Got size_buf [5, 8]: size:1288
Error reading data: Timeout
Finished reading GPS data

Got size_buf [9, 239]: size:2543
Error reading data: Timeout
Finished reading GPS data

Got size_buf [19, 222]: size:5086
Error reading data: Timeout
Finished reading GPS data

```
For `DATA_BUF_SIZE=31` everything works as expected.

## Expected behavior

Reading into a large buffer should output the same result as a smaller buffer.

Here is the output with the buffer size 31:
```
Got size_buf [0, 0]: size:0
Finished reading GPS data

Got size_buf [4, 252]: size:1276
$GNRMC,181453.00,A,3406.41996,N,02306.00679,E,0.009,,250225,,,D,V*1F
$GNVTG,,T,,M,0.009,N,0.017,K,D*37
$GNGGA,181453.00,3406.41996,N,02306.00679,E,2,12,0.62,2.3,M,48.4,M,,*4A
$GNGSA,A,3,27,01,14,08,10,02,32,,,,,,1.14,0.62,0.96,1*05
$GNGSA,A,3,68,82,67,83,,,,,,,,,1.14,0.62,0.96,2*03
$GNGSA,A,3,36,09,25,34,04,02,06,11,30,,,,1.14,0.62,0.96,3*03
$GNGSA,A,3,20,30,41,32,,,,,,,,,1.14,0.62,0.96,4*0E
$GNGSA,A,3,,,,,,,,,,,,,1.14,0.62,0.96,5*0A
$GPGSV,3,1,12,01,40,275,32,02,52,291,28,03,14,215,22,08,84,185,37,1*68
$GPGSV,3,2,12,10,41,049,34,14,13,320,23,16,05,177,19,27,48,135,40,1*6E
$GPGSV,3,3,12,32,30,103,30,36,33,142,41,40,19,118,32,49,40,177,31,1*6C
$GPGSV,1,1,01,23,01,042,,0*52
$GLGSV,2,1,06,67,39,050,23,68,49,125,32,73,20,243,18,75,09,340,15,1*74
$GLGSV,2,2,06,82,28,082,31,83,58,036,26,1*71
$GLGSV,1,1,03,69,15,170,,74,27,291,,84,29,302,,0*4D
$GAGSV,3,1,11,02,42,223,30,04,19,055,20,05,07,157,19,06,26,082,29,7*71
$GAGSV,3,2,11,09,26,108,34,10,00,076,08,11,24,070,34,25,09,179,18,7*77
$GAGSV,3,3,11,30,36,295,28,34,44,278,19,36,69,026,33,7*4D
$GAGSV,1,1,01,27,02,332,,0*40
$GBGSV,2,1,06,20,16,187,,30,79,002,,32,55,133,,36,13,245,,0*72
$GBGSV,2,2,06,36,13,245,,41,34,053,,0*71
$GQGSV,1,1,00,0*64
$GNGLL,3406.41996,N,02306.00679,E,181453.00,A,D*77
2316.02679,E,181453.00,A,DFinished reading GPS data

Got size_buf [4, 252]: size:1276
$GNRMC,181454.00,A,3406.41996,N,02306.00688,E,0.114,,250225,,,D,V*1B
$GNVTG,,T,,M,0.114,N,0.211,K,D*3E
$GNGGA,181454.00,3406.41996,N,02306.00688,E,2,12,0.58,2.4,M,48.4,M,,*4D
$GNGSA,A,3,27,01,14,08,10,02,32,,,,,,1.08,0.58,0.92,1*05
$GNGSA,A,3,68,82,67,83,,,,,,,,,1.08,0.58,0.92,2*03
$GNGSA,A,3,36,09,25,34,04,02,06,11,30,,,,1.08,0.58,0.92,3*03
$GNGSA,A,3,20,30,41,32,,,,,,,,,1.08,0.58,0.92,4*0E
$GNGSA,A,3,,,,,,,,,,,,,1.08,0.58,0.92,5*0A
$GPGSV,3,1,12,01,40,275,32,02,52,291,27,03,14,215,22,08,84,185,36,1*66
$GPGSV,3,2,12,10,41,049,41,14,13,320,22,16,05,177,19,27,48,135,41,1*6C
$GPGSV,3,3,12,32,30,103,23,36,33,142,40,40,19,118,31,49,40,177,31,1*6C
$GPGSV,1,1,01,23,01,042,,0*52
$GLGSV,2,1,06,67,39,050,28,68,49,125,31,73,20,243,20,75,09,340,18,1*7A
$GLGSV,2,2,06,82,28,082,31,83,58,036,26,1*71
$GLGSV,1,1,03,69,15,170,,74,27,291,,84,29,302,,0*4D
$GAGSV,3,1,11,02,42,223,29,04,19,055,23,05,07,157,22,06,26,082,31,7*7B
$GAGSV,3,2,11,09,26,108,34,10,00,076,07,11,24,070,35,25,09,179,21,7*73
$GAGSV,3,3,11,30,36,295,28,34,44,278,23,36,69,026,33,7*44
$GAGSV,1,1,01,27,02,332,,0*40
$GBGSV,2,1,06,20,16,187,,30,79,002,,32,55,133,,36,13,245,,0*72
$GBGSV,2,2,06,36,13,245,,41,34,053,,0*71
$GQGSV,1,1,00,0*64
$GNGLL,3406.41996,N,02306.00688,E,181454.00,A,D*7E
2306.00688,E,181454.00,A,DFinished reading GPS data

Got size_buf [0, 0]: size:0
Finished reading GPS data
```

## Environment

- Target device: esp32 (revision v1.0)
- Crate name and version: esp-hal 0.23.1

bjoernQ, 2025-02-26:
Seems to affect only ESP32 (and maybe ESP32-S2)

bugadani, 2025-02-27:
I strongly suspect this is the same as https://github.com/esp-rs/esp-hal/issues/3167

bjoernQ, 2025-02-27:
> I strongly suspect this is the same as [#3167](https://github.com/esp-rs/esp-hal/issues/3167)

Doesn't seem like that to me - any read larger than 32 bytes triggers it for me (without any other interrupts etc. happening)

Looks like this

[image]

i.e. it generates a start-repeat-condition which it shouldn't

we currently take a new cmd-iterator over and over again (i.e. it starts from the beginning) - reusing the same iterator makes that `sr` go away (but given the max 32 bytes we can read/write at once and we only have 16 commands means we can't get too many bytes in/out)

bjoernQ, 2025-02-27:
I don't have that module (I haven't seen a GPS module which isn't using UART before) but I have a vl53l5-tof sensor which needs some bigger reads (few hundred bytes) and huge writes (32k) - it works fine with ESP32-S3, C3, C6 etc. but NOT with ESP32 and ESP32-S2 .... so I can use that as a kind of reproducer for this issue

So this issue:
- is real
- only affects ESP32 / ESP32-S2

Now that I have some way to reproduce the issue (and some ideas what is happening) I can work on a fix

bjoernQ, 2025-03-04:
Mind trying #3199 to see if it actually fixes this issue?

bjoernQ, 2025-03-05:
> Yes, it fixes the issue except for: [#3199 (comment)](https://github.com/esp-rs/esp-hal/pull/3199#discussion_r1980299083)

Thanks for testing - with the latest commit the remaining problems should be fixed

PR #3199 (2025-03-06): ESP32/ESP32-S2: I2C Workaround
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have [... cut]