esp-rs/esp-hal #3179: `OpenDrain` functionality seems to be broken since `1.0.0-beta.0`

bjoernQ, 2025-02-26:
We removed the explicit `OpenDrainOutput` driver in favor of a config on `Output` (and making it into Flex)

Working with the `OpenDrainOutput` driver on 0.23.1 (connect the two pins)

```rust
    let mut a = OutputOpenDrain::new(peripherals.GPIO6, Level::High, Pull::Up);
    let mut b = OutputOpenDrain::new(peripherals.GPIO7, Level::High, Pull::Up);

    let delay = Delay::new();

    loop {
        println!("high/high");
        a.set_high();
        b.set_high();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("low/high");
        a.set_low();
        b.set_high();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("high/low");
        a.set_high();
        b.set_low();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("low/low");
        a.set_low();
        b.set_low();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!();
    }
```

Non-working on 1.0.0-beta.0  (connect the two pins)

```rust
    let mut a = Output::new(
        peripherals.GPIO6,
        Level::High,
        OutputConfig::default()
            .with_drive_mode(DriveMode::OpenDrain)
            .with_pull(Pull::Up),
    )
    .into_flex();
    let mut b = Output::new(
        peripherals.GPIO7,
        Level::High,
        OutputConfig::default()
            .with_drive_mode(DriveMode::OpenDrain)
            .with_pull(Pull::Up),
    )
    .into_flex();

    let delay = Delay::new();

    loop {
        println!("high/high");
        a.set_high();
        b.set_high();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("low/high");
        a.set_low();
        b.set_high();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("high/low");
        a.set_high();
        b.set_low();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!("low/low");
        a.set_low();
        b.set_low();
        delay.delay_millis(5);
        println!("{} {}", a.is_high(), b.is_high());

        println!();
    }
```

MabezDev, 2025-02-26:
There seems to be a difference in how to set the pin into open drain before https://github.com/esp-rs/esp-hal/pull/3029, we used to call `set_to_open_drain_output` now we just set the `w.pad_driver()` bit in `apply_config` instead. I suppose `set_to_open_drain_output` was doing something we're not doing now. Alternatively the order of operations could matter here.

bjoernQ, 2025-02-26:
I _think_ we can solve it with just some documentation (i.e. an additional step is needed to regain the full functionality of OpenDrainOutput) - let me see

bjoernQ, 2025-02-26:
Yes - no code changes needed

bugadani, 2025-02-27:
> In 1.0.0-beta.0 we never read a high level

Huh did I forget to remove the readback functions?

MabezDev, 2025-02-27:
> > In 1.0.0-beta.0 we never read a high level
> 
> Huh did I forget to remove the readback functions?

Maybe, but according to https://docs.espressif.com/projects/rust/esp-hal/1.0.0-beta.0/esp32c6/esp_hal/gpio/struct.Output.html#method.is_set_high the input stage doesn't need to be enabled, so maybe it's fine?

PR #3181 (2025-02-28): Clarify open drain
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [ ] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have added neces [... cut]
closed 2025-02-28 by MabezDev