esp-rs/esp-hal #3182: RMT - added the ability to use extended RMT memory for a channel

enelson1001, 2025-02-26:
## Motivations

I would like to use all the RMT memory available for one channel.

## Solution

See page 414 of the Technical Reference Manual

## Alternatives

None 

## Additional context

Dominaezzz, 2025-02-28:
> See page 414 of the Technical Reference Manual

For which chip?

enelson1001, 2025-02-28:
The technical manual referenced was for ESP32 but all devices support this option.  
The option is located under the ```RMT_CHnCONF0 register```  and bit name is ``` RMT_MEM_SIZE_CHn```

Here is a summary
## ESP32
- Block Size = 64x32
- RAM size = 512x32
- Channels = 8; can be either TX or RX

## ESP32-S2
- Block size = 64x32
- RAM size = 256x32
- Channels = 4; can be either TX or RX

## ESP32-S3
- Block size = 48x32
- RAM size = 384x32
- Channels = 8; 4TX and 4RX

## ESP32-C2, ESP32-C3, ESP32-C6, ESP32-H2 (not 100% sure on C2)
- Block size = 48x32
- RAM size = 192x32
- Channels = 4; 2TX, 2RX

enelson1001, 2025-03-01:
@Dominaezzz Are you looking at implementing this feature?

Dominaezzz, 2025-03-01:
Nope!

enelson1001, 2025-03-01:
OK I think I will try to implement.

enelson1001, 2025-03-03:
I am getting close to submitting a PR. 

But I have a git question that I hope someone  will answer.
1. I forked esp-hal
2. Cloned the forked repository to my local pc.
3. Checked out a new branch rmt-flt-thrs-fix
4. Made changes to rmt.rs
5. Push changes to my forked copy of esp-hal
6. Created a Pull Request from the Github web site of the forked esp-hal
7. Checked out a new branch rmt-add-memsize-to-configure (but now this branch has changes I made to rmt.rs for the filter threshold fix).
8. Make changes to rmt.rs to add memsize to configure
9. If I push changes to my forked copy of esp-hal can I submit a PR that includes the changes from the previous PR (rmt-flt-thrs-fix)?

Dominaezzz, 2025-03-03:
You can (Git/GitHub will allow you) but imo you shouldn't, makes it harder to review.
If you start your work on a fresh branch, it can be easily reviewed before the existing PR lands (I'm a little surprised it hasn't landed yet).

enelson1001, 2025-03-03:
OK, I think I will wait a couple of days in hopes the RMT filter threshold fix PR will get pulled in.
Thank you for your response.

enelson1001, 2025-03-15:
I have made changes to rmt.rs to implement the ability to use extended RMT memory.  I have tested TX and RX on programs that I have written for ESP32 and ESP32S3 since I have these two boards.  Now I would like to test for for the other boards.  After looking at CONTRIBUTING.md I am still confused at how to run the xtask and HIL to just run test for rmt examples and HIL tests for only RMT  and to run tests for ESP32S2, ESP32C2, ESP32C3, ESP32C6.  Can some show me how to do this?

Once I run these tests I will be able to submit PR.

Will the PR I submit to resolve the issue I opened also resolve PR #3052?

For the PR I submit I was going to use the following commit statement " Resolve #3182 extend RMT memory".  Is this OK and the correct way to submit the PR?

enelson1001, 2025-03-16:
OK solved  with help from esp-rs/element 
```
cargo xtask run-example examples esp32s3 embassy_rmt_rx
```

wisp3rwind, 2025-05-05:
I think this can be closed given #3292 was merged.

closed 2025-05-05 by enelson1001