esp-rs/esp-hal #3183: RMT - filter threshold max value is wrong

enelson1001, 2025-02-26:
## Bug description

The filter threshold is set to 127 while page 418 of the technical reference manual shows this is an 8 bit value.
When you fix the code could you possible add a note that the filter threshold count is based upon the source clock and not the divided clock.

## To Reproduce

1. Set RxChannelConfig;;default().with_filter_threshold(255)
2. Code will crash

The following is the code to poduce error.
```
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::clock::CpuClock;
use esp_println::println;
use log::info;

use esp_hal::{
    gpio::{Level, Output, OutputConfig},
    rmt::{PulseCode, Rmt, RxChannelAsync, RxChannelConfig, RxChannelCreatorAsync},
    time::Rate,
    timer::timg::TimerGroup,
};

extern crate alloc;

#[esp_hal_embassy::main]
async fn main(spawner: Spawner) {
    // generator version: 0.2.2

    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    esp_alloc::heap_allocator!(72 * 1024);

    esp_println::logger::init_logger_from_env();

    let timer0 = esp_hal::timer::timg::TimerGroup::new(peripherals.TIMG1);
    esp_hal_embassy::init(timer0.timer0);

    info!("Embassy initialized!");

    let freq = Rate::from_mhz(80);

    let rmt = Rmt::new(peripherals.RMT, freq).unwrap().into_async();

    let rx_config = RxChannelConfig::default()
        .with_clk_divider(80)
        .with_idle_threshold(700)
        .with_filter_threshold(255); // filter threshold is based on source clock not divided clock so 255 = 3.188uS

    let mut channel = rmt
        .channel7
        .configure(peripherals.GPIO2, rx_config)
        .unwrap();

    let mut data: [u32; 64] = [PulseCode::empty(); 64];

    // TODO: Spawn some tasks
    let _ = spawner;

    loop {
        println!("Hello");

        channel.receive(&mut data).await.unwrap();

        for entry in &data[..data.len()] {
            println!("lvl0 = {:?}   dur0 = {:?}", entry.level1(), entry.length1());
            println!("lvl1 = {:?}   dur1 = {:?}", entry.level2(), entry.length2());
        }

        Timer::after(Duration::from_secs(1)).await;
    }

    // for inspiration have a look at the examples at https://github.com/esp-rs/esp-hal/tree/v0.23.1/examples/src/bin
}
```

## Expected behavior

filter_threshold should accept a value of 255 or less.

## Environment

- Target device: ESP32
- Crate name and version: esp-hal 1.0.0-beta.0

MabezDev, 2025-02-26:
Thanks! This does indeed seem like a bug. Would you mind filing a PR to bump the threshold check?

enelson1001, 2025-02-26:
Is there a document for submitting a PR with steps to follow?

MabezDev, 2025-02-27:
We have a contribution guide, and developer guidelines available here: https://github.com/esp-rs/esp-hal/tree/main/documentation. We also have PR checks in place. I think a change as simple as this should be quite straight-forward (hopefully!).

closed 2025-03-12 by enelson1001