esp-rs/esp-hal #3189: Allow wiping out AES key after usage.

coolwanglu, 2025-02-28:
## Motivations

Currently the AES module [copies the key](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/aes/mod.rs#L78), and it is not wiped out [after usage](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/aes/mod.rs#L172).

## Solution

Use Zeroize on `Key` to automatically wipe it out on drop. This can be guarded by a feature.

Example:

```
#[derive(ZeroizeOnDrop)]
struct Key {}
```

## Alternatives

The AES library does ont make a copy. Instead it takes only a reference and thee caller should wipe it out if necessary.
This changes the API though.

coolwanglu, 2025-02-28:
I could send a PR if this sounds good. Not sure how to test it properly though.

bugadani, 2025-02-28:
It would probably be best if we didn't copy the key just to then copy it into the hardware. Regardless of that, zeroing it on drop sounds reasonable to me, so maybe we should have both?

Dominaezzz, 2025-02-28:
Maybe `Key` should just go away and `process` can just take a slice directly instead.

The fact that `Mode` and `Key` are specified separately makes the `Key` enum feel like boilerplate to me.

This way user can zeroize their slices themselves.