esp-rs/esp-hal #3207: Fix MSRV check

MabezDev, 2025-03-04:
It's currently failing because we need to manage changing toolchain version (as we now run these tests on the self hosted runners). Example job failure: https://github.com/esp-rs/esp-hal/actions/runs/13655133548/job/38172360501?pr=3200

bugadani, 2025-03-04:
Should be fixed by https://github.com/esp-rs/espup/pull/481 ?

MabezDev, 2025-03-04:
Ah, so we need an espup & xtensa-toolchain release, is that right @SergioGasquez?

SergioGasquez, 2025-03-04:
We only need an `espup` release, will prepare it!

SergioGasquez, 2025-03-04:
`espup@0.14.1` is now published and I just rerun the msrv check action and now the `xtensa-toolchain` succeeds https://github.com/esp-rs/esp-hal/actions/runs/13655133548/job/38177931148?pr=3200

closed 2025-03-04 by MabezDev