esp-rs/esp-hal #3235: Low power BLE advertisments

maufl, 2025-03-07:
## Motivations
I would like to sent BLE advertisements (on ESP32-C6) and save as much power as possible. I tried with the ESP IDF in C++ and get much better results if I use dynamic frequency scaling the FreeRTOS tickless idle feature.

## Solution
It would be great if the provided Embassy executor could also go into light sleep if no task is running but wake up for sending BLE advertisements.

## Alternatives
I'm not sure what the alternatives would be.

## Additional context
I tried to read the C++ code and understand what is necessary to implement this feature, but I can't completely wrap my head around it. As far as I understand, it involves multiple features of the ESP: light sleep, modem sleep (which is always disabled in esp-hal), and sleep retention links (that I don't fully understand).

I'm happy to try implementing this if I have enough pointers on how it would theoretically be done. :)

bugadani, 2025-03-10:
It may be possible to write a new executor that puts the chip into light sleep. The only similar implementation I'm aware of if https://github.com/embassy-rs/embassy/blob/7c49f482d71d594d7b48c3393cc98d03a9e7c9e2/embassy-stm32/src/low_power.rs but the sleep APIs in esp-hal are probably not up to the task just yet.

Dominaezzz, 2025-04-17:
> modem sleep (which is always disabled in esp-hal)

Where do you see it disabled? I'm also looking into what it would take to achieve this but on the S3.

bjoernQ, 2025-04-17:
Yes we currently disable BT power-management. For the newer chips it shouldn't be too hard to enable it (but - I tried and failed) - especially for the original ESP32 it doesn't look like fun (not too sure about C3/S3 - but probably not as bad as ESP32 and more work than the newer chips)

Dominaezzz, 2025-04-17:
Are there any docs around how the BT power management works or is it also hush hush?

Mind sharing a link to which line of code (or general area) in esp-wifi I'd have to change to enable power management on the S3? I'm failing to find anything with GitHub search.

I've ordered a 32.768kHz crystal as well to see how low I can get the current consumption to go with ble modem sleep.

bjoernQ, 2025-04-17:
I haven't found any useful documentation unfortunately - so it's looking at the code until it makes sense (or not)

You probably want to look into the bt.c files - especially the cfg-ed "CONFIG_PM_ENABLED" blocks

Someone pioneering here is really appreciated  ❤️

Dominaezzz, 2025-05-05:
Note to self. https://github.com/espressif/esp-idf/blob/3b748a6cb76c2db7c6368a0dea32a88bc58bc44d/components/bt/controller/esp32c3/bt.c#L1181-L1307

The ESP32-S3 (and C3) can enter light sleep whilst BLE is running when using the 32.798kHz XTAL clock.
When using the internal 40MHz RTC clock it can only enter light sleep between connection intervals.

My next task is getting the external RTC clock to work but my not being an EEE isn't helping.

Just leaving this here for future implementers.

bugadani, 2025-05-05:
> The ESP32-S3 (and C3) can enter light sleep whilst BLE is running when using the 32.798kHz XTAL clock.

I've been considering investing time into a clock tree impl, where we would be refcounting which clock is used by which peripheral (right now we just refcount whether a peripheral gets clocks or not, but not which clock is getting routed where), at the very least. We will need something like this before we can think about light sleep, especially light sleep with radio. If you want to work on anything related, I think this would be a good place to start.

closed 2026-09-28 by bjoernQ