esp-rs/esp-hal #3236: Enable octal-psram support for unexpected maker esp32s3 board

maxwen, 2025-03-08:
I have an unexpected maker esp32s3 board
https://esp32s3.com/feathers3.html
This board has 8mb psram

It works fine with quad-psram (giving 2MB psram) but this should support octal-psram - right?
Also giving the espressif datasheet here
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf

But trying to enable octal-psram gives the following error on boot

INFO - vendor id    : 00 (UNKNOWN)
INFO - dev id       : 01 (generation 2)
INFO - density      : 00 (0 Mbit)
INFO - good-die     : 00 (Fail)
INFO - Latency      : 01 (Fixed)
INFO - VCC          : 00 (1.8V)
INFO - SRF          : 00 (Slow Refresh)
INFO - BurstType    : 00 ( Wrap)
INFO - BurstLen     : 00 (16 Byte)
INFO - Readlatency  : 02 (10 cycles@Fixed)
INFO - DriveStrength: 00 (1/1)
WARN - PSRAM ID read error: 0, PSRAM chip not found or not supported, or wrong PSRAM line mode

Given the fact that vendors of esp32 boards do very weired things sometimes
I was wondering if there is any way I can try to narrow down this issue to
get the octal-psram support working

Looking at the octal-psram specific code in psram_init
I found a few comments like 
// CONFIG_SPIRAM_ECC_ENABLE not yet supported
Is the missing support of this board based on this missing things?

If yes is there anything I can do to help here?

bugadani, 2025-03-08:
By eye, that board has at most quad psram. The chip itself has 8 pins on the picture, and you need at least one power and ground pin.

Dominaezzz, 2025-03-08:
> I have an unexpected maker esp32s3 board
> https://esp32s3.com/feathers3.html
> This board has 8mb psram

> 8MB of extra QSPI PSRAM

That board doesn't support octal mode

maxwen, 2025-03-08:
And how can I then access the 8MB as said with quad-psram I only get 2MB

bjoernQ, 2025-03-08:
You can set the PSRAM size in esp-hal's config: https://docs.espressif.com/projects/rust/esp-hal/1.0.0-beta.0/esp32s3/esp_hal/struct.Config.html#method.with_psram

maxwen, 2025-03-08:
Ok now I feel stupid :)

Just for reference if someone else searches for it hot to do it

```
    let mut psram_config = PsramConfig::default();
    psram_config.size = PsramSize::Size(8 * 1024 * 1024);
    let peripherals = esp_hal::init(esp_hal::Config::default().with_psram(psram_config));
```

closed 2025-03-08 by maxwen