esp-rs/esp-hal #3238: The Ieee802154 driver does not use the `PeripheralRef` metaphor

ivmarkov, 2025-03-08:
## Bug description

Since the driver takes the `IEEE802154` and the `RADIO_CLK` peripherals by value, it is not really possible to mutably borrow the peripherals into the radio and then release them once the radio is dropped.

I believe this is just an ommission, as the IEEE802154 Radio is not used that widely yet.

## Expected behavior

Use `impl Peripheral<P = IEEE802154> + 'd`, ...` like everywhere else. :)
 
## Environment

- Target device: esp32c6, esp32h2
- Crate name and version: `main` branch

bugadani, 2025-03-08:
We're simply not there yet

ivmarkov, 2025-03-08:
I know, and `IEEE802154::sreal()` is a good workaround in the meantime.

The bigger trouble is #3239

bugadani, 2025-03-28:
I think I can say this will be "fixed" by https://github.com/esp-rs/esp-hal/pull/3305 - because if that gets merged, we'll not have PeripheralRef anymore :)

closed 2025-05-02 by MabezDev