esp-rs/esp-hal #3248: ESP32S3 i2c timeout

DeppLearning, 2025-03-12:
## Bug description

I am trying to port my gt911 touchscreen driver to a recent esp-hal version, however, I'm seeing I²C Timeouts with previously working code/device.

## To Reproduce

Run this code on an ESP32-S3. I am using the ESP-S3-BOX-3. I am aware that this particular device has problems regarding the wiring of the GT911 which affects the I2C address selection. I know the address that worked previously with the very device I am using, but netherless tried both possible options - same error.

Minimal example: 

```rust
#![no_std]
#![no_main]

use esp_hal::main;
use esp_hal::delay::Delay;
use esp_hal::gpio::Io;
use esp_hal::i2c::master::{BusTimeout, Config, I2c};
use esp_hal::time::Rate;
use embedded_hal::delay::DelayNs;
use esp_backtrace as _;

#[main]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let mut delay = Delay::new();
    let io = Io::new(peripherals.IO_MUX);

    let sda = peripherals.GPIO8;
    let scl = peripherals.GPIO18;

    let mut i2c = I2c::new(peripherals.I2C0, Config::default().with_frequency(Rate::from_hz(100)).with_timeout(BusTimeout::Disabled)).unwrap();
    let reg: u16 = 0x8040;
    i2c.write(0x5d, &reg.to_be_bytes()).unwrap(); // <-- ERROR here
    loop {
        delay.delay_ms(100);
    }
}
```

I am getting this error 

```
[2025-03-12T10:58:09Z INFO ] Serial port: '/dev/ttyACM0'
[2025-03-12T10:58:09Z INFO ] Connecting...
[2025-03-12T10:58:10Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       3c:84:27:05:06:b4
App/part. size:    84,736/1,048,576 bytes, 8.08%
[2025-03-12T10:58:10Z INFO ] Segment at address '0x0' has not changed, skipping write
[2025-03-12T10:58:10Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:00] [========================================]      20/20      0x10000                                                                            [2025-03-12T10:58:11Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378bc6
0x40378bc6 - ESP32Reset
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (24) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (24) boot: compile time Jun  7 2023 08:07:32
I (25) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 40MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c000020 size=019c4h (  6596) map
I (100) esp_image: segment 1: paddr=000119ec vaddr=3fc89414 size=00ab8h (  2744) load
I (108) esp_image: segment 2: paddr=000124ac vaddr=40378000 size=01414h (  5140) load
I (117) esp_image: segment 3: paddr=000138c8 vaddr=00000000 size=0c750h ( 51024) 
I (136) esp_image: segment 4: paddr=00020020 vaddr=42010020 size=04ab0h ( 19120) map
I (142) boot: Loaded app from partition at offset 0x10000
I (142) boot: Disabling RNG early entropy source...

panicked at gt911-examples/examples/bug.rs:24:41:
called `Result::unwrap()` on an `Err` value: Timeout

Backtrace:

0x42014101
core::result::unwrap_failed
    at ??:??
0x42010872
main
    at ??:??
0x42013738
Reset
    at ??:??
0x40378c1e
ESP32Reset
    at ??:??
```
I tried esp-hal versions: "1.0.0-beta.0" as well as latest commit (2cd75e195da5eb5ede9572709c9d0b30bcf424a2). Furthermore I tried extending the two feature gated code blocks that were added here https://github.com/esp-rs/esp-hal/pull/3199/commits/04ee5684eb300564efa6e4bb684d1e8babe171f4 to include the `esp32s3` - same error

## Expected behavior

Well, no timeout error :)

DeppLearning, 2025-03-12:
> Try running the bus clock at 100kHz instead of 100Hz

oh boy sorry, messed up with the minimal example, I mean khz, tried again, still same error.

Dominaezzz, 2025-03-12:
The gt911 does have two addresses, and you can configure which one it uses by pulling the interrupt pin high or low when it powers up. 
Just beware of that. I'd recommend doing a full bus scan to make sure that's not the issue you're having.

DeppLearning, 2025-03-12:
Thx guys for the quick help!

I'm not doing this from scratch but have code that I am merely migrating to the new hal. The code was working before against the same S3-Box. On several of those boxes. Thats how I know the GPIO pins are SDA: 8, SCL: 18. This is the schematic I am using: https://github.com/espressif/esp-box/blob/master/hardware/SCH_ESP32-S3-BOX-3_V1.0/SCH_ESP32-S3-BOX-3-MB_V1.1_20230808.pdf. I am aware of the two addresses, that's why I tested both. As far as I remember the GT911 wiring in the S3-Box is somewhat weird, i.e. the int pin is connected in a way that the esp-hal cannot really make sense of it, i.e. the pin is not really set to 1 or 0, but to 0.5 :| I did a lot of experiments to get around the issue with the prior esp-hal version I was using, to no avail. Sadly I couldn't get hold of the one Box I have in the field, that uses 0x14. Again,  this particular box I am using now always listened to 0x5d, not 0x14. 

I adjusted my minimal code to this:

```rust

#[repr(u8)]
pub enum Address {
    One = 0x5d,
    Two = 0x14,
}

#[main]
fn main() -> ! {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    let mut delay = Delay::new();
    let io = Io::new(peripherals.IO_MUX);

    let sda = peripherals.GPIO8;
    let scl = peripherals.GPIO18;
    let int = Output::new(peripherals.GPIO3, Level::Low, Default::default());
    // let dc = Output::new(peripherals.GPIO4, Level::High, Default::default());
    // let mut rst = Flex::new(peripherals.GPIO48);

    let mut i2c = I2c::new(peripherals.I2C0, Config::default().with_frequency(Rate::from_khz(400)).with_timeout(BusTimeout::Disabled)).unwrap();
    let reg: u16 = 0x8040;
    i2c.write(Address::One as u8, &reg.to_be_bytes()).unwrap();
    loop {
        delay.delay_ms(100);
    }
}

```

I also tried each combination of int pin Level (high / low) and Address (one/two). I even tried different pins for int (gpio 5, not sure where I read this, schematic says gpio3). Same error.

>  you should get an AcknowledgeCheckFailed error I think

This used to be the case indeed whenever I used the wrong address for the wrong box. I just did another I2c scan, there is indeed a device connected on 0x5d, which should be the GT911.

This is the code I am trying to port. This used to work. I have other code (targeting similar or same esp-hal) that is a bit more involved and was using the int pins, that worked too. This worked on the same box I am having problems with now:
https://github.com/DeppLearning/gt911/blob/init2/examples/esp32_poll.rs

DeppLearning, 2025-03-12:
I think I got it.. missed the with_sda and with_scl methods..... sorry for the noise

closed 2025-03-12 by DeppLearning