esp-rs/esp-hal #3251: DMA hangs after it reaches descriptor with `Owner::CPU`

George-Miao, 2025-03-12:
## Bug description

I was testing out the [experimental DMA TX stream buf](https://github.com/Dominaezzz/esp-hal/commit/7ff621e68892c86821b45ec1a5408dd47f2e610c) from @Dominaezzz (I originally found this in a [discussion](https://github.com/esp-rs/esp-hal/discussions/2866)). I have esp32-s3 and a lcd driven by st7701s ([datasheet](https://focuslcds.com/wp-content/uploads/Drivers/ST7701S.pdf)). The screen has a parallel RGB (DPI) with 3-wire SPI interface. The example in the code works well and I do get the expected pure red and blue flashing screen. But whenever I add some time-consuming logic (initialize and update slint, for example) between start of transmission and subsequent data push, the screen turned dark.

I created a [MRE repo](https://github.com/George-Miao/esp-dma-lcd-mre), however it is just for showcasing the code more easily and I don't expect anyone to happen to have the exact same hardware. After some digging, I found all subsequent pushes was eventually early-returned by following code that was supposed to reclaim buffer from DMA:

https://github.com/George-Miao/esp-dma-lcd-mre/blob/eb52c69a275a80a1602ea00fa6e621c72143e05b/src/dma.rs#L298-L300

That's how I come to the conclusion that DMA hangs, since auto write back is turned on but the descriptor was never given back to CPU and no bytes was written to the screen after the delay.

## To Reproduce

1. Run `cargo run --release`
2. The screen should turn red and blue normally
3. Uncomment [`src/main.rs:159`](https://github.com/George-Miao/esp-dma-lcd-mre/blob/eb52c69a275a80a1602ea00fa6e621c72143e05b/src/main.rs#L159), which delays 10ms before the main loop starts
4. DMA hangs and nothing got transmitted to the screen

## Expected behavior

The descriptor should have owner bit set to CPU and data should be transmitted to LCD_CAM normally.

## Environment

- Target device: esp32s3 (revision v0.2)
- Crate name and version: esp-hal 1.0.0-beta.0

Dominaezzz, 2025-03-12:
What's happened here is the DMA got to the end of the linked list and halted. By the time you added more descriptors to the list, the DMA had stopped caring.

LCD_CAM basically doesn't care and will just keep going with zeros. As far as it's concerned, the transfer never ends, so it won't let you know when this happens.

The DMA does have an interrupt for this, OUT_TOTAL_EOF but the catch with it is you have to mark the last descriptor in the list with the EOF flag.

DpiTransfer basically needs to expose the DMA interrupts for you to check for this condition.

Once this condition is detected, there are only two options, restart the transfer or restart just the DMA. In practice you want the former.

George-Miao, 2025-03-13:
I see. Do you know how does esp-idf handle the situation?

Dominaezzz, 2025-03-13:
It restarts the transfer