esp-rs/esp-hal #3252: Unexpected breakage when migrating to 1.0.0-beta.0

dimpolo, 2025-03-12:
When trying to update `esp-hal` to `1.0.0-beta.0` a couple of required changes were difficult to discover because they wern't mentioned in the Migration Guide or the CHANGELOG

This is a summary of what I found:

* some things now live in the `system` module which is wrongly named `cpu` in the CHANGELOG:
  > The Cpu struct and contents of the reset and cpu_control modules have been moved into cpu. (#3099)

* when accessing registers directly I had to add a lot of `.register_block()`s which was difficult to discover
* you can no longer use pin drivers as wakeup sources which was implemented here #2095 and broken by #2419 because `into_ref` now returns `Flex` and not the peripheral

For the last one I haven't found a workaround yet, so this might not be a documentation problem but a bug/feature request

bugadani, 2025-03-12:
Wakeup is more a "we removed something that shouldn't have existed" situation. You can conjure up the pins you need and pass them to the sleep APIs. I'm pretty sure we still have working examples.

bugadani, 2025-03-12:
It shouldn't have existed because pin drivers are not pins, and we should not have built our APIs accordingly. We'll revisit this when we get to designing it properly, and you'll likely have a proper solution at that point. But that's far off.

I ended up conjuring pins in [my project](https://github.com/card-io-ecg/card-io-fw/blob/d46f47356a0cd76f455234fce1e17ae177977f0e/src/main.rs#L296) and it's working adequately. I know it's not elegant. Sleep APIs are not stable for a reason.

MabezDev, 2025-05-02:
> Would if be fair to say then, that it should be possible to reclaim a pin from a pin driver?

This is already possible. At the time of creation of this issue, you can just pass the pin in via `&mut` and then you can just drop the driver to get your pin. On main that's changed just slightly, where now you need to pass the raw pin in by calling `pin.reborrow()` to get that shortened lifetime.

MabezDev, 2025-07-07:
Sorry, I thought I replied 😅. 

My general comment was that `Rtc` is unstable, and it's API is _far_ from perfect. It needs thought and work to improve it, therefore I suggested closing this issue and exploring improving `Rtc`.

> It's rather unergonomic trying to store a pin driver with an attached lifetime

Regarding this comment, It depends what you mean by store, if you mean store in a static, then yes that would be hard. I can maybe see a future where we have a specialized impl for the `'static` lifetime, something like:

```rust
impl OutputPin<'static> {
   pub fn release(self) -> AnyPin {}
}
```

The problem with this (other than it just being an idea, I have no idea if it can work) is that we'd probably have to return `AnyPin`. We _might_ be able to conjure the original pin based on the pin number, but that will likely be a massive match statement. I suggest we explore this in a separate issue.

Finally, the first two points of your issue:

> some things now live in the system module which is wrongly named cpu in the CHANGELOG:

This was a mistake in the changelog, but I don't think this is a point of confusion two releases later.

> when accessing registers directly I had to add a lot of .register_block()s which was difficult to discover

Using the HAL shouldn't require direct register access (and is advised against, unless you know what your doing), and it's an unstable API. Again, there probably should have been a changelog entry for this, but we're not expecting most users to use registers directly. We'll try to do better in the future.

I hope that clears some things up, and please if you're using these rtc APIs consider helping improving them, and in cases where you have to reach for registers directly, let us know :).

bugadani, 2025-07-07:
> Giving users (unstable) access to the PAC should absolutely stay IMO

You are free to depend on the PACs and use them as you wish, but esp-hal can't expose unstable dependencies as part of its API.