esp-rs/esp-hal #3288: Put the critical-section implementation behind an opt-out feature

Dominaezzz, 2025-03-23:
## Motivations

Right now there is no way to provide a bespoke critical-section implementation when using esp-hal, since there can only be one incantation of `critical_section::set_impl!(....);` in a project.

It should be as optional as `esp-hal-embassy` which has a similar "you can only do it once" `embassy_time_driver::time_driver_impl!(...)` incantation.

## Solution

Put it behind a feature.
It's also a good opportunity to stop using it within the hal. Due to the multi-core issues with it.

## Alternatives

None

## Additional context

None at the moment but this will be breaking change to do after stabilisation.

Dominaezzz, 2025-03-28:
I just thought I might want to do this in the future when I get more creative with my runtimes, and it would be a breaking change to make it optional later.

One idea I have for an alternative implementation is removing dual core support from the S3 implementation.
For a lot of projects I only actually use one out of the two cores the S3 has.
For these projects I could use a leaner (but unsafe if I use the other core) CS implementation.

I don't expect esp-hal to provide such a foot-gunny implementation but I want the freedom to use one if I can prove it's safe in my project.

Another idea is, if I setup some kind of rtos, I might want the CS implementation to yield, instead of block.

So I wouldn't say esp-hal's implementation is lacking or has limitations. It's a very nice "It works for 99.99% of people" implementation.

bugadani, 2025-03-28:
We could probably add a `single-core/multi-core` config that removes CpuControl and switches to a single-core optimized CS 👀

> Another idea is, if I setup some kind of rtos, I might want the CS implementation to yield, instead of block.

I think this is a fair point

Dominaezzz, 2025-03-28:
> We could probably add a `single-core/multi-core` config that removes CpuControl and switches to a single-core optimized CS 👀

Could be bypassed with PAC (esp-hal could really use some docs describing when it's okay to use the PACs) and the config option would be affecting the public API.
It's technically another argument for making it a feature I suppose.

I just had another idea for a CS impl. If I'm using both cores but I know that the cores don't need to talk to each other at all. I could have each core use the "single-core optimized CS" and they'll never need to block each other.
If the cores do need to talk, they can talk via atomics or via the some kind of mailbox register 🙂

bjoernQ, 2025-03-28:
> We could probably add a single-core/multi-core config that removes CpuControl and switches to a single-core optimized CS

I have that literally somewhere in my list of ideas 👍

bugadani, 2025-04-01:
> Could be bypassed with PAC

Wouldn't that be `unsafe`? 🤨

bugadani, 2025-04-01:
IMHO when someone sets an "I use my MCU as single-core" config, the unsafe contract starts with the config stating "this chip is now single core, anything else is my fault". This isn't a PAC issue,

MabezDev, 2025-04-01:
In esp-idf they have a FREERTOS_UNICORE config, which I suppose is similar to what we're suggesting here. I'm unsure if there are any optimizations based on that config in esp-idf though.

bugadani, 2025-04-01:
I always read that config as UNICORN, can we have UNICORN, please? 🦄

bjoernQ, 2025-04-02:
Maybe we should open a separate issue to discuss the ~~UNICORN~~ UNICORE config option since this issue will get closed once we merge the associated PR

MabezDev, 2025-04-02:
Good idea, I opened https://github.com/esp-rs/esp-hal/issues/3322

closed 2025-04-02 by bjoernQ