esp-rs/esp-hal #3289: RFC: Provide more power friendly blocking APIs

Dominaezzz, 2025-03-23:
The title doesn't fully capture the ideas I'm writing about here but it at least captures the motive.

As of writing, drivers in the hal come in two modes, `Blocking` or `Async`.
`Blocking` drivers tight loop around an "I'm done" register to block until the work is done.
`Async` drivers set an interrupt handler which does the whole waker stuff and an executor can poll again once it's woken. (The `Blocking` APIs are still available)

I'm currently working on a [smart watch](https://lilygo.cc/products/t-watch-s3) project and power conservation is critical. Whilst using the SPI and I2C drivers in blocking mode, it dawned on me that some extra power could be saved by using `waiti` whilst spinning on the "I'm done" register ([S3 datasheet Section 5.6.2](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf#subsubsection.5.6.2)).

Most of the drivers in hal require going `Async` to take advantage of `waiti` (which comes with it's own set of overheads) and the `Blocking` APIs just spin at max CPU speed, wasting up to 20mA of current. This leaves no way for users to actually block in a power efficient manner.

Now having stated the problem, the vague proposal follows:

`Blocking` APIs in the hal should use `waiti` (or `wfi`) where possible.

`waiti <priority level>` works by putting the CPU in a low power state and waiting until an interrupt handler above `<priority level>` fires.

This means that the hal has to set up an interrupt handler to be able to block using `waiti`.
`Blocking` drivers currently expose interrupts and handlers to users, which would allow users to set `waiti` unfriendly handlers.

To solve this problem, an additional `DriverMode` can be added called `Polling`.
- `Polling`: Users have full control of how to wait for the driver to finish and they have access to interrupts.
- `Blocking`: Users get a driver that does `waiti` and sets a compatible interrupt handler.
- `Async`: Users get a driver that sets an interrupt handler to wake some waker and does `async` stuff.

`Blocking` and `Async` could probably be merged into a single mode but I don't want to focus too closely on that here.

`Polling` could be considered a "low level" mode, where users could do `waiti` themselves at a priority level of their choice, or do a tight loop or move it into a static to progress the transfer in an interrupt perhaps.

bugadani, 2025-03-24:
A bit branstormy but:

As the interrupt APIs aren't stable currently, this isn't an API question so much as an implementation question. The problem is, if we start changing interrupt APIs to start sleeping, we'll not be able to add interrupt function to the drivers - because the bits will be in use, and users mustn't be able to manipulate them.

We **could** limit the APIs to "listen only" where the driver would provide the status bits to the interrupt handler, but not the option to clear them. Users would receive the first (of each) event they subscribe to, but they couldn't break the driver. I'm not completely sure how useful that would be.

In a `Polling` mode I'm not sure we should provide the current set of blocking (or interrupt) APIs at all. We should instead have `start_transmit` etc, and make the user poll for completion, or continue with the operation. We would likely have to return Busy errors from `start_*` functions, but we can implement this in a way where the driver internally locks the peripheral, and it can also call the interrupt handlers (if they are set) with a reference to the driver itself, so the user can decide how to continue without manually polling. Or without moving the driver into a `static`. Or without locking in the interrupt handler, although that is risky and we may need to do it in the driver anyway. I believe there are interesting possibilities here, at least on paper.

Dominaezzz, 2025-03-24:
> I'm not completely sure how useful that would be.

Even with the current APIs I'm not sure how useful interrupts are when the drivers blocks till completion anyway.

> I'm not sure we should provide the current set of blocking (or interrupt) APIs at all

100% agree and this is what I have in mind. 

I'm thinking of something similar to the DMA APIs (sans the moving) where you get a transfer object with a `is_done` method on it. (This avoids have to return busy errors at least)

(Fwiw the DMA APIs aren't affected by this issue, as users can decide how to wait)

bjoernQ, 2025-03-24:
> I'm thinking of something similar to the DMA APIs (sans the moving) where you get a transfer object with a is_done method on it. (This avoids have to return busy errors at least)

I think I like this idea