esp-rs/esp-hal #3291: I2C transaction_async gets ArbitrationLost error when preempted while an equivalent write_async operation does not

ajwock, 2025-03-23:
## Bug description

I'm using the esp_rust devboard, so this bug should be reproducible on common hardware. https://github.com/esp-rs/esp-rust-board

With what appears to be two equivalent ways of expressing an async burst write transaction with the esp hal, one of them gets an ArbitrationLost error when preempted.  Without the preemption which is imposed by async wifi startup tasks in the example below, both versions work.

The transaction_async version ends up issuing two separate transactions via hardware.  I'm not sure why exactly arbitration gets lost here, but this same issue may be possible for other ops like write_read_async where the driver trades control with hardware multiple times for a given transaction.

Regarding hal source, it doesn't appear the i2c master driver takes full advantage of the hardware command fifo for the write_read and transaction cases.  I suspect this costs reliability under cpu load.

## To Reproduce

These are two different versions of the functions which I wrote for burst writes to the icm42670 imu peripheral.  I wrote the original version to avoid some copy logic.
```rust
     async fn burst_write_regs(&mut self, start_address: u8, reg_vals: &[u8]) {
     self.comm.transaction_async(ACCEL_ADDRESS, &mut [Operation::Write(&[start_address]), Operation::Write(reg_vals)])
             .await
             .expect(format!("Failed to burst write vals {:?} to registers starting at {}", reg_vals, start_address).as_str());
     }
```
```rust
    async fn burst_write_regs(&mut self, start_address: u8, reg_vals: &[u8]) {
        let mut to_write = SmallVec::<[u8; 32]>::new();
        to_write.push(start_address);
        to_write.extend_from_slice(reg_vals);
        self.comm.write_async(ACCEL_ADDRESS, to_write.as_slice())
            .await
            .expect(format!("Failed to burst write vals {:?} to registers starting at {}", reg_vals, start_address).as_str());
    }
```

Here is a somewhat reduced version of the failure.  Though preemption bugs aren't often hard failures, this seems to do it 100% of the time on my board.

Excuse the overall length- I was expanding upon the embassy AP example while also trying to configure the IMU.  It may be possible to get it to preempt with far less excess code.

icm42670.rs
```rust
use esp_hal::i2c::master::{I2c, Operation};
use esp_hal::Async;
use alloc::format;
use esp_println::println;

// 7 bit address of the accelerometer
pub(crate) const ACCEL_ADDRESS: u8 = 0b1101000;

pub(crate) struct Icm42670<'a> {
    comm: I2c<'a, Async>,
}

impl<'a> Icm42670<'a> {
    pub(crate) fn new(i2c: I2c<'a, Async>) -> Self {
        Self {
            comm: i2c,
        }
    }

    async fn write_reg(&mut self, reg_address: u8, val: u8) {
        self.comm.write_async(ACCEL_ADDRESS, &[reg_address, val])
            .await
            .expect(format!("Failed to write val {} to register {}", val, reg_address).as_str());
    }

    async fn burst_write_regs(&mut self, start_address: u8, reg_vals: &[u8]) {
        self.comm.transaction_async(ACCEL_ADDRESS, &mut [Operation::Write(&[start_address]), Operation::Write(reg_vals)])
            .await
            .expect(format!("Failed to burst write vals {:?} to registers starting at {}", reg_vals, start_address).as_str());
    }

    // Registers have an 8-bit address
    async fn read_reg(&mut self, reg_address: u8) -> u8 {
[... 214 lines cut ...]

#[embassy_executor::task]
async fn net_task(mut runner: Runner<'static, WifiDevice<'static>>) {
    runner.run().await
}
```

## Expected behavior

Not this:
====================== PANIC ======================
panicked at src/icm42670.rs:29:14:
Failed to burst write vals [70, 71, 64, 2, 2] to registers starting at 32: ArbitrationLost

## Environment

- Target device:
Chip type:         esp32c3 (revision v0.4)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       a0:85:e3:06:67:cc

(It's the esp-rust devboard)
- Crate name and version: esp-hal = { version = "1.0.0-beta.0", features = ["esp32c3", "unstable"] }

bugadani, 2025-03-24:
Do you experience the same issue with the current git `main`?

bjoernQ, 2025-03-24:
I just copied to code to a fresh project and it's working fine on a Rust-Board v1.0 for me

```
Power on ICM42670
start connection task
Device capabilities: Ok(EnumSet(Client | AccessPoint | Mixed))
Starting wifi
Wifi started!
ICM42680 powered on, verifying identity
Got id: 0x67
Configuring accelerometer and gyro
ICM42680 configured
```

(with the released HAL and also with main)

ajwock, 2025-03-27:
Apologies for the delay.  Unfortunately this issue does still occur at git main for me.  I'll poke around more and see if I can get more consistent replication instructions.

ajwock, 2025-03-27:
I also encountered this occasionally with write_read_async occasionally.

I did fork the repo and make a change that implements write_read_async in one hardware-driven transaction at the Driver level, which resolved the issue and fully unblocks my use case if I avoid the same transaction_async expression as above.  I can work on productionizing that diff if it sounds worth accepting.

closed 2025-05-08 by MabezDev

bugadani, 2025-05-08:
I think I found the root cause of this issue but feel free to reopen if you still experience it.