esp-rs/esp-hal #3298: Error[E0428]: the name 'generate_memory_extras' is defined multiple times

yusefpasha, 2025-03-25:
**Bug description**
When using the esp-hal crate version 1.0.0-beta.0 in a modular Rust workspace project targeting multiple ESP32 microcontrollers, the build fails with a compilation error: error[E0428]: the name 'generate_memory_extras' is defined multiple times. This error occurs in the build.rs file of the esp-hal crate, specifically at lines 256 and 274, indicating a duplicate definition of the generate_memory_extras function. This prevents successful compilation of the project despite correctly configured feature flags and dependencies.

## To Reproduce [](https://github.com/yusefpasha/SampleModular.git)
1. Create a Rust workspace with the following structure:
```
root/
├── Cargo.toml
├── app/
├── core/
├── driver/
│   ├── rgb_led/
│   │   └── Cargo.toml
│   ├── seven_segment/
├── microcontroller_esp32/
└── microcontroller_esp32c3/
```

2. Configure the root Cargo.toml as follows:
```
[workspace]
resolver = "2"
members = [
    "app",
    "core",
    "driver/rgb_led",
    "driver/seven_segment",
    "microcontroller_esp32",
    "microcontroller_esp32c3",
]

[workspace.package]
version = "0.0.1"
authors = ["Babaee"]
description = "A short description of my package"
documentation = "https://example.com/bar"

[workspace.dependencies]
tokio = { version = "1.44.1", default-features = false }
esp-hal = { version = "1.0.0-beta.0" }
smart-leds = { version = "0.4.0" }
esp-hal-smartled = { git = "https://github.com/taorye/esp-hal-community", rev = "1f8ab9a", default-features = false }
tm1637-embedded-hal = { version = "0.5.0-alpha.1", default-features = false }
esp-alloc = { version = "0.7.0", default-features = false }
esp-backtrace = { version = "0.15.0", default-features = false, features = ["exception-handler", "panic-handler", "println"] }
esp-println = { version = "0.13.0", default-features = false, features = ["log"] }
```

3. Configure the driver/rgb_led/Cargo.toml as follows:
```
[package]
name = "rgb_led"
version = "0.1.0"
edition = "2024"

[features]
esp32 = ["esp-hal-smartled/esp32", "esp-hal/esp32"]
esp32c2 = ["esp-hal/esp32c2"]
esp32c3 = ["esp-hal-smartled/esp32c3", "esp-hal/esp32c3"]
esp32c6 = ["esp-hal-smartled/esp32c6", "esp-hal_esp32c6"]
esp32h2 = ["esp-hal-smartled/esp32h2", "esp-hal_esp32h2"]
esp32s2 = ["esp-hal-smartled_esp32s2", "esp-hal_esp32s2"]
esp32s3 = ["esp-hal-smartled/esp32s3", "esp-hal/esp32s3"]

[dependencies]
esp-hal = { workspace = true }
smart-leds = { workspace = true }
esp-hal-smartled = { workspace = true }
```

4. Observe the build failure with the following error:
```
error[E0428]: the name `generate_memory_extras` is defined multiple times
   --> C:\Users\yusef\.cargo\registry\src\index.crates.io-1949cf8c6b5b557f\esp-hal-1.0.0-beta.0\build.rs:274:1
    |
256 | fn generate_memory_extras() -> Vec<u8> {
    | -------------------------------------- previous definition of the value `generate_memory_extras` here
...
274 | fn generate_memory_extras() -> Vec<u8> {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `generate_memory_extras` redefined here
    |
    = note: `generate_memory_extras` must be defined only once in the value namespace of this module
```

## Expected behavior

The project should compile successfully without errors related to duplicate function definitions in the esp-hal crate's build.rs file. The generate_memory_extras function should be defined only once, allowing the build process to complete and enabling the use of esp-hal features for ESP32 variants as specified in the rgb_led module.

## Environment

Target device: ESP32 (and variants like ESP32-C3, configurable via features)
Crate name and version: esp-hal 1.0.0-beta.0

bugadani, 2025-03-25:
Your workspace `Cargo.toml` has `esp-hal = { version = "0.23.1" }`. Did you link the correct repository?

yusefpasha, 2025-03-25:
@bugadani Yeah, I connected it correctly. I have to use this version for the things I need.  
The project had no issues under normal conditions, but when I used a workspace, this problem occurred.  
Feature detection is not working correctly.

closed 2025-05-02 by MabezDev