esp-rs/esp-hal #3308: SpiDmaBus half_duplex_write Unsupported DataMode

zzccppp, 2025-03-28:
## Bug description

the combinaton of datamode below seems not working: 

```rust 
let res = self.spi.half_duplex_write(
            DataMode::Quad,
            Command::_8Bit(0x32, DataMode::Single),
            Address::_24Bit(0x2C << 8, DataMode::Single),
            0,
            color,);
```

## Expected behavior

It should write the data normally.

## Environment

- Target device:
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
- Crate name and version: [esp-hal 1.0.0-beta.0]

zzccppp, 2025-03-28:
I didn't notice the latest changelog where spi::DataMode changed the meaning of DataMode::Single - it now means 3-wire SPI (using one data line). Use DataMode::SingleTwoDataLines to get the previous behavior.
The name SingleTwoDataLines is indeed confusing, as it doesn't include the Quad (four-line) scenario in its naming.

bugadani, 2025-03-28:
I think we have this backwards, though.

bjoernQ, 2025-04-02:
I think the source of confusion is the changed meaning of `Single` - my original plan was to keep `Single` as 4-wire and have `Single3Wire` but we changed the definition a couple of times in the PR

I agree that for address and cmd-phase it shouldn't matter since only MOSI/SIO0 is used to clock out data and nothing is read - so we can relax that check which technically means we just don't differentiate there between the two single-modes.

_Probably_ the check is technically not needed (but better to double check) and was introduced for "philosophical" reasons - e.g. should we allow `Single/Single/SingleTwoLines` or `SingleTwoLines/SingleTwoLines/Single` (or any other combination and only set SIO according to the data-phase)? Technically fine but looks odd, doesn't it?

bugadani, 2025-04-02:
> should we allow Single/Single/SingleTwoLines or SingleTwoLines/SingleTwoLines/Single (or any other combination and only set SIO according to the data-phase)? Technically fine but looks odd, doesn't it?

Odd or not, the hardware may actually allow it. And if the user wants 4-wire half duplex with some odd number of written bits, they'll need to be using the command phase to achieve it.

PR #3325 (2025-04-03): HD SPI: Remove restrictive three-wire check
This technically fixes #3308 although it's unclear to me what the hardware would do for cases like `(Command::_8Bit(0x32, DataMode::Single), Address::_24Bit(0x2C << 8, DataMode::Single), DataMode::SingleTwoDataLines)`, so they are not tested, but not rejected either in this PR. That might change either here or later if I manage to figure out a test setup.
closed 2025-04-03 by MabezDev