esp-rs/esp-hal #3309: documentation website has not been updated for 1.0.0-beta.0

Be-ing, 2025-03-30:
## Bug description

The documentation at https://docs.esp-rs.org/esp-hal/ for esp-hal still shows the documentation for 0.23.1. This caused me confusion trying to use 1.0.0-beta.0 because it took me a bit to figure out the API I was trying to use had changed between the versions and I was looking at old documentation.

SergioGasquez, 2025-03-31:
You can find the latest documentation in https://docs.espressif.com/projects/rust/index.html

bjoernQ, 2025-03-31:
It should also be fixed with the next release #3174

Be-ing, 2025-03-31:
> You can find the latest documentation in https://docs.espressif.com/projects/rust/index.html

Ah, I had the old URL bookmarked. I updated my bookmark.

jessebraham, 2025-03-31:
We should probably redirect the old docs site to the new one.

Be-ing, 2025-03-31:
It's nice to have all the esp-* crates documented in one place now.

MabezDev, 2025-03-31:
I've added a permanent redirect from docs.esp-rs.org to the new docs landing page. This might 24-48 hours to propagate.

bjoernQ, 2025-03-31:
> I've added a permanent redirect from docs.esp-rs.org to the new docs landing page. This might 24-48 hours to propagate.

Oh .... now the book is gone: https://docs.esp-rs.org/book/

MabezDev, 2025-04-01:
> > I've added a permanent redirect from docs.esp-rs.org to the new docs landing page. This might 24-48 hours to propagate.
> 
> Oh .... now the book is gone: [docs.esp-rs.org/book](https://docs.esp-rs.org/book/)

I've undone the redirect (I didn't even work in the end, just hung in the browser for me), and added back the DNS records for Github pages, so the book etc should come back shortly.

MabezDev, 2025-04-01:
I've removed the domain from esp-rs.github.io repo, so http://esp-rs.github.io/book still works for now.

It seems the domain level redirect with the domain provider we're using is pretty limited. I'm going to file some other issues  to resolve this properly.

Be-ing, 2025-04-03:
Do you plan to move the book to docs.espressif.com as well?

SergioGasquez, 2025-04-03:
> Do you plan to move the book to https://docs.espressif.com as well?

Yes, its already moved! https://docs.espressif.com/projects/rust/book/

bjoernQ, 2025-04-03:
Just updated the link in the repo description

Be-ing, 2025-04-03:
It would be nice to put a link to the book on https://docs.espressif.com/projects/rust/index.html to have all the documentation accessible from one URL.

bjoernQ, 2025-07-04:
I guess I resolved this issue - try these links which should now redirect to reasonable locations

Some example links

https://docs.esp-rs.org/esp-hal
https://docs.esp-rs.org/esp-hal/esp-hal
https://docs.esp-rs.org/esp-hal/esp-wifi
https://docs.esp-rs.org/book
https://docs.esp-rs.org/book/installation/index.html
https://docs.esp-rs.org/no_std-training
https://docs.esp-rs.org/no_std-training/03_1_panic.html

bjoernQ, 2025-07-04:
@MabezDev do you consider this done? Anything missing?

MabezDev, 2025-07-04:
I think that's everything, thank you!

closed 2025-07-04 by MabezDev

Daenges, 2026-03-09:
@bjoernQ the website seems hijacked. Your provided links redirect to a very shady website that is not the book.

> I guess I resolved this issue - try these links which should now redirect to reasonable locations
>
> Some example links
>
> https://docs.esp-rs.org/esp-hal https://docs.esp-rs.org/esp-hal/esp-hal [...]