esp-rs/esp-hal #3311: Implement Rust Crypto traits for Crypto Accelerators

bjoernQ, 2025-03-31:
We already have the digest traits for SHA but ideally, we will provide more

See https://github.com/RustCrypto/traits

 - AES
 - SHA 🆗
 - RSA
 - ECC
 - HMAC

Frostie314159, 2025-06-11:
The only thing that I can come up with off the top of my head is to have some crate put the peripheral drivers in statics. The shared structs could then contain the internal state of the accelerator for that instance, which would have to be saved and restored for every operation.
I'm pretty sure that this would nullify all speed advantage that crypto accelerators have.

bugadani, 2025-06-13:
As far as I understand, the crypto peripherals are clocked lower than the CPU is. I remember reading somewhere that they can be slower than software. But this doesn't change two things:

- software is not responsible for the security aspects of the algorithm
- the CPU is free to work on other tasks, if the traits offered async APIs (but still, an interrupt wouldn't pause a calculation)

> I could also imagine to just add a way to save/restore the context to the low-level drivers contained in the HAL and have a separate crate like esp-hal-crypto implementing the Crypto-traits

> The only thing that I can come up with off the top of my head is to have some crate put the peripheral drivers in statics.

A separate crate might be a good way forward.

closed 2026-09-07 by MabezDev