esp-rs/esp-hal #3327: Handling of different bootloaders / bootloader versions

bjoernQ, 2025-04-03:
# Current Situation

- the HAL won't include an app-descriptor which is needed by later versions of the esp-idf 2nd stage bootloader
- we want to support more bootloaders in future (namely MCUboot, maybe direct-boot again?)
- future bootloaders might require tweaking of linker scripts / configurations

# Problems

- other than esp-idf we don't build the bootloader together with the firmware
   -  for convenience espflash comes with pre-built bootloaders - these are bundled with the binary and might get updated when a new version is released
- supporting another bootloader might require a different design of the linker scripts
- future versions of a supported bootloader might require tweaks to the linker scripts

# Things discussed

- consider warning the user when using a bundled pre-built bootloader to avoid getting surprised when a project doesn't boot anymore (after just updating espflash)
- move linker scripts to the bootloader-support crates for more flexibility

MabezDev, 2025-04-03:
Adding 1.0 blocker, just so we can make a decision (not necessarily implement anything) before the full 1.0 release.

Dominaezzz, 2025-04-03:
Is [this list](https://github.com/esp-rs/esp-hal/blob/6d781b0cdf749f5cf6f1ecefb85180e88b2e360c/esp-hal/src/system.rs#L126-L136) bootloader specific? If so it's another problem to add to the list.

MabezDev, 2025-04-03:
> Is [this list](https://github.com/esp-rs/esp-hal/blob/6d781b0cdf749f5cf6f1ecefb85180e88b2e360c/esp-hal/src/system.rs#L126-L136) bootloader specific? If so it's another problem to add to the list.

No, that list is for esp-hal to keep enabled. Disabling things like usb-serial-jtag is a bad idea :D.

bugadani, 2025-05-27:
> but that just sucks

at one point we have to take a stance between "we have esp-generate to do it" vs "we cram everything into esp-hal because it'd be hard to use"

bjoernQ, 2025-05-27:
> The problem is linker scripts aren't really open to this kind of "addon" behavior ...

The support-crates can contribute to the linker search path just fine (we do that in a couple of places) - so if we agree on a common name for these scripts we can just include them and it will be found (if the crate is included)

We will have the same "problem" with the future esp-rom-crate ( #3389 ) but I think we can just assume it to be available (since esp-hal will depend on it) - if someone wants to use it w/o esp-hal users need to manually include them

MabezDev, 2025-06-23:
One idea I'm toying with right now is pulling out the LD stuff from esp-hal. A new crate, `esp-ld` or `esp-hal-ld` would have all the linker files for all use cases. Bootloader crates can then depend on it, and enable the relevant features. The linker search params and final link.x file can be generated once all at build time, instead of having to worry about fallible INCLUDE's.

It's a bit weird having an essentially empty crate, but I think it would work.

MabezDev, 2025-06-23:
> I don't find "let's call the relevant file bootloader.ld" that bad

If a user depends on two bootloader crates (we have no way of knowing with this approach), the bootloader.ld that's used is whatever cargo decides to build last. To be fair, that's the only downside I see with this simple approach. In fact, I think each bootloader crate could check for the existence of the bootloader.ld file and detect multiple that way 🤔.

I'm curious what @bjoernQ found though 👀

bugadani, 2025-06-23:
Why would anyone depend on two bootloader crates at the same time?

MabezDev, 2025-06-23:
> Why would anyone depend on two bootloader crates at the same time?

Not intentionally, I hope, but it's something that might happen. We need to detect it because debugging that would be insane.

MabezDev, 2026-11-10:
The linker scripts now live in the bootloader support crate with hooks for other bootloaders, which was the plan here, and direct boot is tracked in #6560: https://github.com/esp-rs/esp-hal/commit/56a449ec365a3462f4fafb7167550fc7da23cf41

closed 2026-11-10 by MabezDev