esp-rs/esp-hal #3337: Broken link in `esp_hal_procmacros` documentation

VorpalBlade, 2025-04-04:
## Bug description

https://docs.espressif.com/projects/rust/esp-hal-procmacros/0.17.0/esp_hal_procmacros/index.html (which appears to be the most recent version as of writing this) has two broken links in the intro text. Under "Key Components" it links to "interrupt" and "main". Both of these are 500 Internal Server Error

## To Reproduce

1. Visit https://docs.espressif.com/projects/rust/esp-hal-procmacros/0.17.0/esp_hal_procmacros/index.html
2. Click links.

## Expected behavior

Links work.

## Environment

- Target device: N/A
- Crate name and version: esp_hal_procmacros 0.17.0

jessebraham, 2025-04-07:
> Also: You should probably give 404, not 500, on missing pages. That seems misconfigured.

Yeah, that's definitely the wrong thing to do. @MabezDev any idea why this is happening?

MabezDev, 2025-04-07:
So I figured out what's going on here.

We're not building esp-hal-procmacros with the right feature set, but our docs link checking isn't catching this because it was written to link directly to the file, not a syntactic rustdoc link, which 404's when it's not there.

Two things to fix here:

1) Merging the 2-3 ways to specify features to enable for a package to just one place in the xtask
2) Stop 404's from redirecting endlessly, and timing out with a 500

I'm addressing the former currently.

closed 2025-04-14 by MabezDev

VorpalBlade, 2025-04-14:
The server is still returning 500 instead of 404 though. So perhaps this was closed prematurely?

MabezDev, 2025-04-15:
I just redeployed the docs for the proc macros package, this should resolve the issue.