esp-rs/esp-hal #3376: esp-alloc - switch from `linked_list_allocator` to `embedded-alloc`

ivmarkov, 2025-04-12:
## Motivations

No particular problem with `linked_list_allocator`, yet perhaps switching to the `embedded-alloc` facade would allow folks extra flexibility, in that it provides the [TLSF](https://docs.rs/rlsf/latest/rlsf/) allocator as an option as well.

Benefits would be more predictable performance (TLSF has constant-time allocs) and _maybe_ less heap fragmentation.

(ESP-IDF uses TLSF too.)

## Solution

As per the motivations.

## Alternatives

Leave it as is, hard-coded to llff.

bjoernQ, 2025-04-14:
I also thought of configurable algorithms - maybe per memory region.

TIL there is https://crates.io/crates/rlsf/0.2.1 which doesn't come with the 62KiB allocation limit like https://crates.io/crates/tlsf

ivmarkov, 2025-04-14:
> I also thought of configurable algorithms - maybe per memory region.

Per memory region would be the "Rolls-Royce" option though benefit perhaps not completely obvious, but then why not indeed?

> TIL there is https://crates.io/crates/rlsf/0.2.1 which doesn't come with the 62KiB allocation limit like https://crates.io/crates/tlsf

Yes. Rlsf is precisely what `embedded-alloc` uses when it's `tlsf` feature is enabled.

bjoernQ, 2025-09-17:
Unassigned myself - we'll likely want to resurrect #3950

PR #4130 (2025-09-17): Use TLSF by default
Attempt 2 (previous one: #3950 - github doesn't let me reopen that PR)

Closes #3376