esp-rs/esp-hal #3381: HIL: Disconnect GPIO4 and GPIO5 on the C6 device

Dominaezzz, 2025-04-14:
Please disconnect GPIO4 and GPIO5 on the C6 device used for running HIL tests. They are not currently used in any tests.

I pretty much ran out of pins for https://github.com/esp-rs/esp-hal/pull/3339 and had to use the questionable GPIO10 and GPIO11.

Wasn't sure were to ask for this (since #1334 was closed) so I've started here. I figure any objections can be discussed here.

MabezDev, 2025-04-22:
Thanks, we'll disconnect these pins now. IIRC they were connected prior to the interconnect module existing.

playfulFence, 2025-04-22:
Done 

[image]

playfulFence, 2025-04-22:
Just random [CI run](https://github.com/esp-rs/esp-hal/actions/runs/14593353329/job/40933766315) went just fine after this (just to check), closing the issue.

bugadani, 2025-04-22:
This also needs to be documented in hil-test/README.md, and we should at the same time remove other unneeded connections as well.

closed 2025-04-22 by bjoernQ