esp-rs/esp-hal #3385: Support of exporting esp_hal::main through crate without main application dependent on esp-hal

georgik, 2025-04-14:
## Motivations

Let's have an application which is using intermediate board support crate which deals with HW. The crate has dependency on esp_hal and is able to access `esp_hal::main`. It should be able to re-export the main function, so the parent application can use `#[main]`.

This behavior is currently not possible, because re-export of `main` proc macro will cause the parent application requiring `esp_hal`.

```
error[E0433]: failed to resolve: use of undeclared crate or module `esp_hal`
  --> demos/printerdemo_mcu/main.rs:36:1
   |
36 | #[mcu_board_support::entry]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^ use of undeclared crate or module `esp_hal`
```

## Solution

The woarkaround is to add dependency on esp_hal also to parent application.

## Additional context

https://github.com/slint-ui/slint/pull/8112/#issuecomment-2801767477

## How to simulate

```sh
git clone https://github.com/georgik/slint.git --branch feature/esp-hal-1.0.0
cd slint
CARGO_PROFILE_RELEASE_OPT_LEVEL=s cargo +esp run -p printerdemo_mcu --target xtensa-esp32s3-none-elf --no-default-features --features=esp32-s3-box-3 --release --config examples/mcu-board-support/esp32_s3_box_3/cargo-config.toml
```

Then remove dependency in `demos/printerdemo_mcu/Cargo.toml` - `esp-hal`, try build again.
The build will fail.

Slint is using here, the following `entry` which should be alias to `main`:
```
#[mcu_board_support::entry]
fn main() -> ! {
```

bugadani, 2025-04-14:
The solution is to implement an entry macro in `mcu_support_crate`. It needs to output code that refers to `mcu_support_crate`, and `mcu_support_crate` needs to re-export esp_hal. Then maybe we can change our main macro to optionally take a proxy crate (like #[main(mcu_support_crate)]` - which would generate code like `::mcu_support_crate::_export::esp_hal::blabla` instead of `::esp_hal::blabla`), so that the macro in `mcu_support_crate` would substitute in this attribute.

tronical, 2025-04-14:
Yep, solved it that way now. @ogoffart suggested basically to use `use mcu_board_support::prelude::*;` and the crate re-exports `esp_hal` in the prelude.

closed 2025-04-15 by MabezDev