esp-rs/esp-hal #3400: Is there an example of using ESP32-C6 with PARL_IO in i8080 mode to drive an LCD?

rehe0x, 2025-04-21:
## Motivations

I’m trying to use the ESP32-C6’s PARL_IO peripheral in i8080 mode to control an LCD, and I’d like to find an example or demo showing how to do this.
## Solution

## Alternatives

## Additional context

Dominaezzz, 2025-04-21:
> I’d like to find an example or demo showing how to do this.

There isn't one in this repo. See https://github.com/esp-rs/esp-hal/issues/940

> ESP32-C6’s PARL_IO peripheral in i8080 mode

Strictly speaking, there isn't a "i8080 mode" on the PARL_IO. It can just be used as is.

It seems esp-idf [doesn't have it](https://github.com/espressif/esp-idf/commit/e890b4bd7eba26a8de2010cba96a0893e1560ece) either.

Which devkit are you using for this? I might give it another try.

rehe0x, 2025-04-21:
I am using ESP32-C6-WROOM-1 with nostd. The peripheral is an ST7796 LCD in 8-bit parallel mode. I followed the example in the PARL_IO documentation, but I am unsure how to configure the WR (Write Enable) pin. I tried mapping the WR pin to ClkOutPin and manually controlling the DC pin. Data transmission works fine, but the LCD still does not function properly. Unfortunately, I don’t have a logic analyzer, so it’s hard to troubleshoot.

Dominaezzz, 2025-04-21:
> I tried mapping the WR pin to ClkOutPin and manually controlling the DC pin. Data transmission works fine, but the LCD still does not function properly.

Looks right to me. That's how I'd do it.

What about the CS pin?

Just in case you didn't know, the clock pin is free running.

rehe0x, 2025-04-21:
> CS 引脚呢？
> 
> 以防万一您不知道，clock pin 是自由运行的。

Thanks for the reminder, I think I understand now. Currently, CS is held low, and I should set CS low to select the device right before sending data. I'll give it another try tomorrow.

MabezDev, 2025-04-22:
Please share your code if you get this working, just as a reference for others :)

rehe0x, 2025-04-22:
> Please share your code if you get this working, just as a reference for others :)

I haven’t succeeded yet. The simplified code is as follows. I’m not sure if the configuration is correct — I’ve tried various methods, but none of them worked.

rehe0x, 2025-04-22:
```rust 
let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    let rst: Output<'_> = Output::new(peripherals.GPIO19, Level::Low, OutputConfig::default());
    let mut bl: Output<'_> = Output::new(peripherals.GPIO18, Level::Low, OutputConfig::default());

    let mut rs_dc: Output<'_> = Output::new(peripherals.GPIO23, Level::High, OutputConfig::default());
    let mut cs: Output<'_> = Output::new(peripherals.GPIO15, Level::High, OutputConfig::default());

    let dma_tx_buf = dma_tx_buffer!(32000).unwrap();
    let dma_channel = peripherals.DMA_CH0;

    let mut tx_pins = TxEightBits::new(
        peripherals.GPIO4, // lcd db0
        peripherals.GPIO5, // lcd db1
        peripherals.GPIO6, // lcd db2
        peripherals.GPIO7, // lcd db3
        peripherals.GPIO0, // lcd db4
        peripherals.GPIO1, // lcd db5
        peripherals.GPIO2, // lcd db6
        peripherals.GPIO3, // lcd db7
    );

    let parl_io = ParlIoTxOnly::new(peripherals.PARL_IO, dma_channel, Rate::from_mhz(1)).unwrap();

    // lcd wr
    let mut clock_pin = ClkOutPin::new(peripherals.GPIO22);

    let parl_io_tx = parl_io
        .tx
        .with_config(
            &mut tx_pins,
            &mut clock_pin,
            0,
            SampleEdge::Normal,
            BitPackOrder::Msb,
        )
        .unwrap();
    
    // lcd command
    dma_tx_buf.as_mut_slice()[0] = 0x11;
    cs.set_low();
    rs_dc.set_low();
    let transfer = parl_io_tx.write(1, dma_tx_buf).unwrap();
    (_, parl_io_tx, dma_tx_buf) = transfer.wait();
    rs_dc.set_high();
    cs.set_high();

    // lcd data
    dma_tx_buf.as_mut_slice()[0] = 1;
    dma_tx_buf.as_mut_slice()[1] = 2;
    cs.set_low();
    let transfer = parl_io_tx.write(2, dma_tx_buf).unwrap();
    (_, parl_io_tx, dma_tx_buf) = transfer.wait();
    cs.set_high();

rehe0x, 2025-04-23:
> I strongly recommend using TxPinConfigWithValidPin::new(tx_pins, cs.split().1.inverted()) for the CS pin

After checking the logic analyzer, I found that the configuration
TxPinConfigWithValidPin::new(tx_pins, cs.split().1.inverted())
for the CS pin is very critical.

> It'd be a good idea to get a logic analyzer. You never know if you're dealing with a hardware, driver or firmware bug, and it's quite frustrating to find out you've been looking in the wrong place.

I bought a logic analyzer and did a simple test. I found that the DMA buffer doesn’t get updated — it always contains the data set during the first transmission. Here’s the code:

```rust
  dma_tx_buf.fill(&[0x01,0x02]);
  let transfer = parl_io_tx.write(2, dma_tx_buf).unwrap();
  let  (_, parl_io_tx,  mut dma_tx_buf) = transfer.wait();

  dma_tx_buf.
  dma_tx_buf.fill(&[0x22]);
  let transfer = parl_io_tx.write(1, dma_tx_buf).unwrap();
  let  (_, parl_io_tx,  mut dma_tx_buf) = transfer.wait();

```

The second transmission should send 0x22, but according to the logic analyzer, it still sends 0x01, which was used in the first call.

Dominaezzz, 2025-04-23:
Sounds like a driver bug, perhaps we're not resetting the FIFO properly.

Dominaezzz, 2025-05-01:
Classic. I added a loop to the 8-bit HIL test and the 2nd iteration fails...

I'm a little surprised no one has ran into this issue yet.

Dominaezzz, 2025-05-01:
@rehe0x please give https://github.com/esp-rs/esp-hal/pull/3448 a try

rehe0x, 2025-05-06:
> Classic. I added a loop to the 8-bit HIL test and the 2nd iteration fails...
> 
> I'm a little surprised no one has ran into this issue yet.

I ran the test again and can confirm that I also see a failure on the second transmission.