esp-rs/esp-hal #3401: Is there any support for Classic Bluetooth / SPP ?

0-Sacha, 2025-04-21:
## Motivations

I am new to bluetooth so I probably be wrong.

I am using an esp32 (first version) and I would like to connect to [Stellarium Telescope Control](https://stellarium-labs.com/telescope-control-in-stellarium-mobile-plus/). Their control method said they only support bluetooth 2.0, using SPP profile.

I first tried using `trouble` but figured out that is not gonna work as `SPP` is for Classic Bluetooth and that `trouble` is only a BLE stack.

## Solution

I have not been able to find any no-std crate that can provide Classic Bluetooth functionality, and from what I have understand, `esp-hal/esp-wifi` only support Bluetooth BLE ?

## Alternatives

- The only alternative should be `esp-idf`, but my project is using `embassy` with no-std. 
- I could switch back to C++ on the framework Arduino.
- Or use an serial bluetooth controller but that is sad using an esp32 then...

## Additional context

bjoernQ, 2025-04-22:
> I have not been able to find any no-std crate that can provide Classic Bluetooth functionality, and from what I have understand, esp-hal/esp-wifi only support Bluetooth BLE ?

Finding / creating such a stack would be a start. In addition there are are very few places in esp-wifi where we currently hard-code to initialize the controller in BLE mode (instead of classic mode). (e.g. here: https://github.com/esp-rs/esp-hal/blob/bc35b6f2fd46cebf3545a9afe8a50922d516d610/esp-wifi/src/ble/btdm.rs#L438C14-L438C31 )

But t.b.h. I never even tried to enable/use BT Classic

0-Sacha, 2025-04-23:
Ok thanks !
I will try to make the hal for classic bluetooth.

I am new to esp_hal and bluetooth so any clues are welcome 😉

bjoernQ, 2025-04-24:
You probably want to start with the Bluetooth Core Specification to see how things should work and use `BleConnector` to talk to the controller

KOEGlike, 2025-05-12:
I'm also happy to help with this

ljodal, 2025-08-12:
I've been working on a Rust wrapper for [btstack](https://github.com/bluekitchen/btstack) to use Bluetooth Classic in a Rust based project. I currently have a somewhat functional proof of concept for Bluetooth Low Energy, but I seemed to have hit a wall when trying to enable Bluetooth Classic features. As is pointed out here, it seems like the controller is only initialized for BLE usage. Would it be possible to make changes to at least allow the controller to be used in dual-mode?

Here's what I've found so far:

1. Enable the controller in dual-mode:
  https://github.com/esp-rs/esp-hal/blob/ca4e28da4a905ecaede96d0ea03e26890f13a2fa/esp-radio/src/ble/btdm.rs#L448
  
2. Maybe the init call also needs to be changed?
  https://github.com/esp-rs/esp-hal/blob/ca4e28da4a905ecaede96d0ea03e26890f13a2fa/esp-radio/src/ble/btdm.rs#L418-L422

3. Call `config_bt_funcs_reset ` in the `patch_apply` function:
  https://github.com/esp-rs/esp-hal/blob/ca4e28da4a905ecaede96d0ea03e26890f13a2fa/esp-radio/src/ble/os_adapter_esp32.rs#L189-L201

4. At least `mode` in the `esp_bt_controller_config_t` struct has to be changed, maybe more things?
  https://github.com/esp-rs/esp-hal/blob/ca4e28da4a905ecaede96d0ea03e26890f13a2fa/esp-radio/src/ble/os_adapter_esp32.rs#L292-L322

With these changes locally I'm able to get the device to advertise itself over classic, but I'm getting a timeout error when trying to pair from iOS. I'm not super familiar with Rust nor Bluetooth, so I'm not entirely sure where the problem occurs but it appears to be outside of my control. I'm getting the incoming connection HCI event and btstack responds to it with an accept connection request. After this I get a timeout failure event back from the controller after ~90 seconds, while the esp-idf based project with btstack I get a success response within milliseconds.

I'd be happy to share my btstack project, but it's currently in a very rough state.

ljodal, 2025-08-12:
Yes, the ESP32 series support dual-mode BLE and BR/EDR. I have working code based on the esp-idf framework that does exactly what I want to achieve in my Rust-based project

ljodal, 2025-08-25:
I've attempted to make the changes I've found so far here: https://github.com/esp-rs/esp-hal/compare/main...ljodal:bluetooth-dual-mode

This is kinda-sorta working, while at the same time not working. The device shows up under bluetooth settings on iOS, but when trying to pair it fails and I'm not quite sure I understand why. There's the HCI log from my pairing attempt, if that helps anyone pinpoint what could be wrong:

```
[00:36:54.625] EVT <= 04 0A 5D 55 E0 CF 66 FC 0C 02 7A 01
[00:36:54.626] INF -- hci.c.3807: Connection_incoming: FC:66:CF:E0:55:5D, type 1
[00:36:54.627] INF -- hci.c.329: create_connection_for_addr FC:66:CF:E0:55:5D, type fd
[00:36:54.635] INF -- hci.c.7296: sending hci_accept_connection_request
[00:36:54.641] CMD => 09 04 07 5D 55 E0 CF 66 FC 01
[00:36:54.655] EVT <= 6E 00
[00:36:54.656] EVT <= 0F 04 00 05 09 04
[00:36:54.675] EVT <= 03 0B 00 80 00 5D 55 E0 CF 66 FC 01 00
[00:36:54.676] INF -- hci.c.3840: Connection_complete (status=0) FC:66:CF:E0:55:5D
[00:36:54.677] INF -- hci.c.3877: New connection: handle 128, FC:66:CF:E0:55:5D
[00:36:54.685] INF -- hci.c.8232: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:36:54.690] EVT <= 61 01 01
[00:36:54.694] EVT <= CD 09 80 00 00 5D 55 E0 CF 66 FC
[00:36:54.698] INF -- att_server.c.404: SM_EVENT_IDENTITY_RESOLVING_STARTED
[00:36:54.705] INF -- sm.c.2328: LE Device Lookup: not found
[00:36:54.710] EVT <= CE 09 80 00 00 5D 55 E0 CF 66 FC
[00:36:54.715] INF -- att_server.c.424: SM_EVENT_IDENTITY_RESOLVING_FAILED
[00:36:54.722] EVT <= 05 04 00 80 00 08
[00:36:54.725] INF -- ATT write callback: con_handle=0, att_handle=0x0000, transaction_mode=3, offset=0, buffer_size=0
[00:36:54.735] WRN -- ATT write callback called with null-buffer
[00:36:54.741] INF -- hci.c.1284: Connection closed: handle 0x80, FC:66:CF:E0:55:5D
[00:36:54.748] INF -- hci.c.8232: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:36:54.754] EVT <= 61 01 00
[00:37:15.855] EVT <= 04 0A 5D 55 E0 CF 66 FC 0C 02 7A 01
[00:37:15.857] INF -- hci.c.3807: Connection_incoming: FC:66:CF:E0:55:5D, type 1
[00:37:15.857] INF -- hci.c.329: create_connection_for_addr FC:66:CF:E0:55:5D, type fd
[00:37:15.865] INF -- hci.c.7296: sending hci_accept_connection_request
[00:37:15.871] CMD => 09 04 07 5D 55 E0 CF 66 FC 01
[00:37:15.885] EVT <= 6E 00
[00:37:15.886] EVT <= 0F 04 00 05 09 04
[00:37:53.335] EVT <= 03 0B 08 81 00 5D 55 E0 CF 66 FC 01 00
[00:37:53.336] INF -- hci.c.3840: Connection_complete (status=8) FC:66:CF:E0:55:5D
[00:37:53.337] INF -- hci.c.2631: Outgoing connection to FC:66:CF:E0:55:5D failed
```

If anyone's able to help work on this I'd really appreciate it. I'd love to write my firmware in rust, but I need bluetooth classic (a2dp), so I kinda have to get this working. I'm still working on cleaning up my rust-wrapper for btstack where I'm using this fork of esp-hal/esp-radio. Hope to maybe be able to share that in a few days, which might help to figure out why it doesn't work.