esp-rs/esp-hal #3424: Migration guide step for new idf bootloaders

MabezDev, 2025-04-25:
Related to the discussion in https://github.com/esp-rs/espflash/pull/857 we'll need to document this migration step.

bjoernQ, 2025-06-18:
I guess this is just a reminder here and the migration guide is for espflash? Do we still need it if we remind users via https://github.com/esp-rs/espflash/pull/872 ? (and addressing https://github.com/esp-rs/espflash/issues/886 )

But having some migration guide wouldn't hurt anyways I guess

MabezDev, 2025-06-23:
Good point. I think with https://github.com/esp-rs/espflash/pull/872 we don't need to worry about this anymore. I create a draft release for esp-hal, and added some notes/todo's there so we don't forget.

Closing.

closed 2025-06-23 by MabezDev