esp-rs/esp-hal #3426: esp-storage NVS API

Erik1000, 2025-04-25:
An NVS API similar to the one in esp-idf-svc would be great to store persistant data in flash. 
I found #190 and looked into esp-storage which implements the basic flash access but a higher level api would be very nice. To be more precise, I am especially missing something like [`EspNVs`](https://docs.esp-rs.org/esp-idf-svc/esp_idf_svc/nvs/struct.EspNvs.html) with the different `set_x` and `get_x` methods.

bugadani, 2025-04-25:
For the time being, you should be able to use [sequential-storage](https://docs.rs/sequential-storage/latest/sequential_storage/) or other crates that are built on top of embedded-storage

AnthonyGrondin, 2025-04-26:
Unless I'm wrong, `sequential-storage` seems to depend on `embedded-storage-async`. And I think Async support isn't implemented for `esp-storage`

Also, I believe, further down the line, it would be nice to have a good documentation in the book about storing data in NVS, and how to use different higher level crates to interact with the flash memory (`sequential-storage`, `ekv`, `NorFS`, etc.)

Erik1000, 2025-04-26:
Yea, I just looked into it and `sequential-storage` indeed uses async so it cannot be used.

bugadani, 2025-04-26:
The [BlockingAsync adapter](https://github.com/embassy-rs/embassy/blob/main/embassy-embedded-hal/src/adapter/blocking_async.rs) should work for that

Erik1000, 2025-04-26:
Well it would but `esp_storage::Flash` does not implement `embedded_storage::nor_flash::NorFlash` which is required by `sequential-storage`

bugadani, 2025-04-26:
[It will in the next release](https://github.com/esp-rs/esp-hal/blob/ab73f43d7d05292277502bf23a1261facb03669a/esp-storage/src/nor_flash.rs#L158)

Erik1000, 2025-04-26:
Can confirm that the workaround with sequential storage works!

bjoernQ, 2025-11-20:
Just found https://crates.io/crates/esp-nvs

Haven't checked it but looks promising